nightscout/AndroidAPS #1062: Error message during eros Pod activation

Thomas-Wolff, 2021-12-11:
Hello everyone, once again I had problems with an eros pod activation this night.
Deactivation of the Pod went very well. When activating the new Pod, after transferring the basal rate to the Pod, I got the following error message 'Checking activation step failed. Please try again'.
I then tried again, but this also failed with the same message.
I then cancelled the activation and tried again with the same Pod. The screen then immediately jumped to telling me to attach the pod to my body, which I did and the activation went through without a problem. Pod tab looked good afterwards and pump ran immediately without error.
Attached is a screenshot and the log file from the activation at 01:16 tonight.

Hardware:
Samsung Galaxy A40 Android 11
AAPS 2.8.2.14-dev Build: 057d072e8
OL FW 3.2
[image]
[AndroidAPS._2021-12-11_00-57-03_.2[1].zip](https://github.com/nightscout/AndroidAPS/files/7696792/AndroidAPS._2021-12-11_00-57-03_.2.1.zip)

Greetings Thomas

MilosKozak, 2021-12-11:
it was timeout while communication
`01:17:50.311 [RxComputationThreadPool-4] E/PUMPBTCOMM: [RFSpy.writeToData():236]: writeToData: RileyLink reports timeout
01:17:50.312 [RxComputationThreadPool-4] W/PUMPBTCOMM: [RileyLinkCommunicationManager.sendAndListen():90]: isDeviceReachable. Response is invalid ! [noResponseFromRileyLink=false, interrupted=false, timeout=true, unknownCommand=false, invalidParam=false]
01:17:50.313 [RxComputationThreadPool-4] D/PUMPBTCOMM: [OmnipodRileyLinkCommunicationManager.exchangePackets():353]: Ignoring exception in exchangePackets: RileyLinkCommunicationException: Timeout
`

closed 2021-12-11 by MilosKozak

Thomas-Wolff, 2021-12-11:
Hello Milos, thanks for the information, I had something like that in mind but unfortunately I couldn't understand it.

Greetings Thomas