nightscout/AndroidAPS #1066: 3.0. Automatization and temporary profile switch

Korostensky, 2021-12-12:
New feature in automatization
--------------
Now isn't possible switch profile with duration without reset autosens. Thank you 👍.

MilosKozak, 2021-12-12:
it should work with latest dev. test it or wait for beta 10.
create new issue if it will not work you

Korostensky, 2021-12-15:
[image]

Korostensky, 2021-12-15:
Zdravím, nevím, jakým způsobem se to mělo vyřešit. Zkoušel jsem v 3.0-beta10.

Stále ale není možnost přidat časové období v automazaci ke změně profilu (dočasná změna profilu). Viz PrintScreeny. 

Zkoušel jsem i možnost u % přepnutí profilu na jinou hodnotu než 70-130% a tam je stále hlídání s invalidní akcí při spuštění. To ale předpokládám nebylo tak zamýšleno, jako to šlo ve verzi 2.8.2 jednoduše pomocí přepsání. Proto tam ta validace je a nelze ji obejít. To je správně.

Nevim, jestli dělám něco špatně nebo jsem to nepochopil.
Děkuji za informaci. 

[image]
[image]
[image]

MilosKozak, 2026-10-11:
Automation can now start a temporary profile switch. Use the "Profile percentage" action with a duration, or "Run scene" with a scene that switches to a named profile for the duration of the scene. A temporary switch does not reset autosens.

closed 2026-10-11 by MilosKozak