nightscout/AndroidAPS #1067: zero value for SMB recorded.

pbloom209, 2021-12-12:
Dana I
Beta 3 ver 9
I have noticed that on the home screen it shows the triangle for SMB being delivered.
When checking in treatments they are zero.  This is confirmed on the pump checking the history.

Please see screenshot
[image]
[image]
s

MilosKozak, 2021-12-12:
version?
logs?

pbloom209, 2021-12-14:
Logs as requested

[AndroidAPS._2021-12-13_02-39-49_.2.zip](https://github.com/nightscout/AndroidAPS/files/7708010/AndroidAPS._2021-12-13_02-39-49_.2.zip)

pbloom209, 2021-12-14:
this issue is continuing to occur.  I cannot see any pattern of events that is triggering this.

MilosKozak, 2021-12-14:
log is 10:27 - 15:24
i need a time to search for

pbloom209, 2021-12-14:
[image]

New screenshot.

With new log.

pbloom209, 2021-12-14:
[AndroidAPS (1).log](https://github.com/nightscout/AndroidAPS/files/7711387/AndroidAPS.1.log)

pbloom209, 2021-12-14:
also note it does not seem to change the SMB from 0.1

pbloom209, 2021-12-14:
search for 22:24 and 22:29

pbloom209, 2021-12-14:
i have just noticed the pattern when AAPS changes the Temp basal rate it also posts in the treatments the 0 SMB and adds the triangle to the home screen

pbloom209, 2021-12-14:
it just did this pattern again.  So the changing of the Basal creates a zero 
smb

MilosKozak, 2021-12-14:
what is your bolus step in doctor's menu?

MilosKozak, 2021-12-14:
could you try it manually on your pump?
start bolus of 0.05 without TBR and with 40% TRB running

pbloom209, 2021-12-14:
I will change the mininum in the doctors menu to 0.05

pbloom209, 2021-12-14:
tried 0.05 with both basal on the pump no problem

pbloom209, 2021-12-14:
They are now showing 0.05 in pump history when I look at it in AAPS

pbloom209, 2021-12-14:
I think it is fixed.  now i have changed to 0.05 in Doctors menu.
Tx

closed 2021-12-14 by pbloom209