nightscout/AndroidAPS #1069: Pixel Pro 6 Android 12 no longer connects to Dash

tswire, 2021-12-14:
Starts at 16:53:48.880

BT connection permission issue.

[AndroidAPS_LOG_1639432456805.log.zip](https://github.com/nightscout/AndroidAPS/files/7707741/AndroidAPS_LOG_1639432456805.log.zip)

AndroidAPS 2.8.2.17-dev
Build: c21b4beeb-2021.12.13-13:54
Remote: https://github.com/nightscout/AndroidAPS.git
Flavor: fullrelease
Nightscout version: UNKNOWN
Engineering mode enabled

MilosKozak, 2021-12-14:
did you try to activate dash with 2.8.2.17 or only continue session?

tswire, 2021-12-14:
Continue a session only. I will be changing pods today so can try to activate a new one?

mick711, 2021-12-14:
Yes you can like i say on discord chanel, update doesn't work but clean install and activate new pod work

MilosKozak, 2021-12-14:
i don't have android 12 phone. Unable to test it

tswire, 2021-12-14:
Ok, I will switch pods today on .17 and activate and go from there.  I wonder if subsequent dev updates will behave the same?  Or is this something we think dev 17 only requires a new pod activation to grant permission then it will behave like before (in session dev updates will be ok??)

parapenT1sta, 2021-12-14:
Just a note, I remember when I start testing Android S (12 beta version) and xDrip didn't work at all with Dexcom but was able to work with MiaoMiao and Bubble, first try to fix was adding the permission request on xDrip https://github.com/NightscoutFoundation/xDrip/pull/1734 but didn't work. The only solution that worked was this one https://github.com/NightscoutFoundation/xDrip/pull/1723.

This cannot be related at all to this issue.

tswire, 2021-12-15:
Adding to this, Dev 18 is unable to be tested as it crashes on Pixel Pro 6.  Both in place and fresh install crashes and must go back to Dev 14 and MUST reinstall fresh Dev 14 to get BT connection back.

mick711, 2021-12-15:
parapenT1sta i use xDrip with G6 on my Zenfone 8 to Android 12 and it work fine but i use last Nightly .
For the moment i use with my Dash Pod v17 Built c21b4beeb... i can't be update AAPS but start a new pod . i don't update for the moment to v18 tomorow i need to change pod so i try to update before .

tswire, 2021-12-15:
> parapenT1sta i use xDrip with G6 on my Zenfone 8 to Android 12 and it work fine but i use last Nightly . For the moment i use with my Dash Pod v17 Built [c21b4be](https://github.com/nightscout/AndroidAPS/commit/c21b4beeba4c53b083e6e05f1e2c03abc5400457)... i can't be update AAPS but start a new pod . i don't update for the moment to v18 tomorow i need to change pod so i try to update before .

Please let us know if Dev 18 works.  It crashes for me on both in place and fresh install.  I do not have a copy of Dev 17 so must stay on Dev 14 for everything to work.  I hope @MilosKozak can either delay Dev 14 expiry (tomorrow) or help test new Dev 19 that works

tswire, 2021-12-15:
It seems most recent commits for Dev 18 has fixed this.  not sure if each new version will prompt a new BT permission but it forced me to request that permission and then connected to Dash pod.  Good to close this?

erobinson, 2021-12-16:
FYI - I updated to dev 18 yesterday and it asked for additional permissions but the connections were totally fine to dash and I had no issues.

mick711, 2021-12-16:
Update to dev 18 today built 35ded... work fine no additional request for me but I update 17 c21... Yes I think issue can be closed

closed 2021-12-16 by MilosKozak