nightscout/AndroidAPS #1072: [3.0 Beta 10] Crashes after start

jroth-haj, 2021-12-14:
Beta 10 in Pixel 5 (Android 12) 
Unable to initialize after fresh Installation or Update.
No logs available.
No configuration. It crashes directly after start.

parapenT1sta, 2021-12-15:
For me either.
Android 12
Google Pixel 4a 5G

Philoul, 2021-12-15:
Do you both have an Insight Pump ? (I made a PR...)

parapenT1sta, 2021-12-15:
> Do you both have an Insight Pump ? (If yes I made this PR #1073...)

Combo.

Maybe due these changes: https://github.com/nightscout/AndroidAPS/commit/31a2a03efb359bd8b687c09c21b064dfb5e87f4c or/and https://github.com/nightscout/AndroidAPS/commit/7cd22ef5e4db538583315a765deefcb73464768e

happygirlnyc70, 2021-12-15:
Same issue  using both beta 10 and dev 17
Pixel 4a 5g
Android 12
Omnipod Eros

StephenBrown2, 2021-12-15:
Same with Beta10, downgrading to Beta9 seemed to work fine, but wouldn't connect to Eros. Trying to Scan from AndroidAPS then caused it to crash repeatedly. Logs attached.
[AndroidAPS_LOG_1639543424780.log.zip](https://github.com/nightscout/AndroidAPS/files/7716539/AndroidAPS_LOG_1639543424780.log.zip)
Pixel 5
Android 12
Omnipod Eros

Alamo04, 2021-12-15:
Confirmed with
Pixel 5
Android 12
Combo

jroth-haj, 2021-12-15:
Its not pump specific. The crash is also at fresh installation.

parapenT1sta, 2021-12-15:
> Do you both have an Insight Pump ? (If yes I made this PR #1073...)

THANKS A LOT! :)

Just built https://github.com/nightscout/AndroidAPS/commit/afea158d0e13d668106ffe2812c693e763dd205a and it's working perfectly fine!
Issues with Combo are also fixed and I assume with other pumps too!

parapenT1sta, 2021-12-15:
@MilosKozak Can you release Beta 11 with this new fix?

andybluenoes, 2021-12-15:
Same issue here with Beta 10
App crashes on startup
Pixel 4
Android 12
Dash

KittDoesCode, 2021-12-15:
Crashes for me too. Start, get a couple of (expected) pop-up alerts, and then crash.

Pixel 5, Android 12
Omnipod Dash + Dexcom G6

Cleared files/cache. Same crash. Uninstalled and installed fresh. Same crashing.

Alamo04, 2021-12-15:
@parapenT1sta 
The fix was merged into Beta10.

parapenT1sta, 2021-12-15:
Beta 11 was released with the fix.

KittDoesCode, 2021-12-15:
I confirm that Beta 11 built successfully, installed (Pixel 5, Android 12), and does not crash on opening. I haven't gotten through any further testing yet. Thank you.

elgrupo123, 2021-12-15:
Here also. Beta11 + NSClient, Android 12, Pixel 4a5G, Dash. 
Everything is running smoothly. . Thank you!!!

Alamo04, 2021-12-15:
Everything is working fine now, with Beta 11.

mxsrm, 2021-12-15:
@MilosKozak 
Beta 11 crashs immediately after start for me on Pixel 6 with DanaRS. Beta 9 no problems at all.

I see a warning "Application needs bluetooth permission" and then the application crashs. All possible permissions have been given (see screenshot).

[image]

[crash.log](https://github.com/nightscout/AndroidAPS/files/7723075/crash.log)

jroth-haj, 2021-12-15:
I had the same.
I deinstalled and have done a new installation.
Then I started many times and it crashes after one second
Then I was fast enough to click to the BT permissions.
After that, it was working well, still without the new Install Wizard.

Then I got to the details of the app in Android and cleared all data. After that it was starting with the new Installation wizard.

Right now, its working well.

mxsrm, 2021-12-15:
Yes I also managed to click the dismiss button after multiple tries, after that it started working (crashed once more though I think).

StephenBrown2, 2021-12-16:
I've just checked out beta11, but Android Studio is giving me this error:
```
Unsupported Modules Detected: Compilation is not supported for following modules: AndroidAPS. Unfortunately you can't have non-Gradle Java modules and Android-Gradle modules in one project.
```

pbloom209, 2021-12-16:
I am also have the same issue with Dana I Beta 11.

StephenBrown2, 2021-12-16:
> clearing the Bluetooth storage

How did you clear bluetooth storage? Do you mean the app storage?

pbloom209, 2021-12-16:
Yes app, selected system apps located bluetooth storage then clear.

But it has not solved the issue.  Happened again today.  So I exited the AAPS. reopened and it connected.

jroth-haj, 2021-12-16:
Could this issue bei closed?
After clearing all caches and data, and with a new clean installation all is working well. So I think it's no longer an bug.

StephenBrown2, 2021-12-16:
Upgrades are still a problem though, so the permission could be requested in a better way, maybe?

mxsrm, 2021-12-17:
> Could this issue bei closed? After clearing all caches and data, and with a new clean installation all is working well. So I think it's no longer an bug.

This is not a real solution because people don't want to lose all settings every time they upgrade. This is still a bug.

jroth-haj, 2021-12-17:
@mxsrm have you tryed an update from the current release Version 2.8.2?
 https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html

"Data is not migrated to new database. Do not complain, it’s so huge change so it’s simply not possible. Thus after update IOB, COB, treatments etc. will be cleared. You have to create new profile switch and start with zero IOB and COB. Plan the update carefully!!! Best in situation without active insulin and carbs"

This update will not bei like other updates before.

I'll test it. If I don't get problems, I prefer to close this issue.

happygirlnyc70, 2021-12-17:
I can confirm I was able to update to beta 11 with an active pod.  I got a red bar request for Bluetooth permission.  I did change pump driver to virtual and then back to Eros after update an orange link connected in less than 15 seconds
Pixel 4a 5g Android 12

mxsrm, 2021-12-17:
> @mxsrm have you tryed an update from the current release Version 2.8.2? https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html
>
> "Data is not migrated to new database. Do [...]

I was using beta 9 before.

jroth-haj, 2021-12-17:
> I was using beta 9 before.

This will not bei the standard situation for standard user / non beta testers.

tanja3981, 2021-12-17:
I can't solve this with DanaRS.
The app crashes faster than I would be able to confirm any bluetooth request. So the reason should be: why does it crash on requesting bluetooth?
This is the exception I saw:
`2021-12-17 22:05:22.791 20623-20680/info.nightscout.androidaps I/WM-WorkerWrapper: Worker result FAILURE for Work [ id=470ff17a-d3c5-4176-8fb5-863482579f58, tags={ info.nightscout.androidaps.plugins.general.nsclient.NSClientAddUpdateWorker } ]
2021-12-17 22:05:23.003 20623-20623/info.nightscout.androidaps W/cout.androidap: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
2021-12-17 22:05:23.004 20623-20623/info.nightscout.androidaps W/cout.androidap: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (unsupported, reflection, allowed)
2021-12-17 22:05:23.011 20623-20668/info.nightscout.androidaps I/FA: Tag Manager is not found and thus will not be used
2021-12-17 22:05:23.293 20623-20770/info.nightscout.androidaps I/AdrenoGLES-0: QUALCOMM build                   : 781e7d0, I46ff5fc46f
    Build Date                       : 12/01/20
    OpenGL ES Shader Compiler Version: EV031.31.04.01
    Local Branch                     : QPR1
    Remote Branch                    : 
    Remote Branch                    : 
    Reconstruct Branch               : 
2021-12-17 22:05:23.293 20623-20770/info.nightscout.androidaps I/AdrenoGLES-0: Build Config                     : C P 11.0.1 AArch64
2021-12-17 22:05:23.293 20623-20770/info.nightscout.androidaps I/AdrenoGLES-0: Driver Path                      : /vendor/lib64/egl/libGLESv2_adreno.so
2021-12-17 22:05:23.301 20623-20770/info.nightscout.androidaps I/AdrenoGLES-0: PFP: 0x016ee190, ME: 0x00000000
2021-12-17 22:05:23.300 20623-20623/info.nightscout.androidaps W/RenderThread: type=1400 audit(0.0:37014): avc: denied { search } for name="5000000.qcom,kgsl-3d0" dev="sysfs" ino=30120 scontext=u:r:untrusted_app_27:s0:c114,c257,c512,c768 tcontext=u:object_r:sysfs_msm_subsys:s0 tclass=dir permissive=0 app=info.nightscout.androidaps
2021-12-17 22:05:23.307 20623-20770/info.nightscout.androidaps W/AdrenoUtils: <ReadGpuID_from_sysfs:197>: Failed to open /sys/class/kgsl/kgsl-3d0/gpu_model
2021-12-17 22:05:23.307 20623-20770/info.nightscout.androidaps W/AdrenoUtils: <ReadGpuID:221>: Failed to read chip ID from gpu_model. Fallback to use the GSL path
2021-12-17 22:05:23.321 20623-20681/info.nightscout.androidaps I/WM-WorkerWrapper: Worker result SUCCESS for Work [ id=5986aa45-781d-49d1-a22f-c44b6538b1ba, tags={ info.nightscout.androidaps.receivers.KeepAliveReceiver$KeepAliveWorker } ]
2021-12-17 22:05:23.327 20623-20788/info.nightscout.androidaps I/Gralloc4: mapper 4.x is not supported
2021-12-17 22:05:23.328 20623-20788/info.nightscout.androidaps W/Gralloc3: mapper 3.x is not supported
2021-12-17 22:05:23.332 20623-20788/info.nightscout.androidaps W/Gralloc4: allocator 4.x is not supported
2021-12-17 22:05:23.332 20623-20788/info.nightscout.androidaps W/Gralloc3: allocator 3.x is not supported
2021-12-17 22:05:23.455 20623-20683/info.nightscout.androidaps I/WM-WorkerWrapper: Worker result SUCCESS for Work [ id=f539d00e-f84b-4bd0-88c5-d6791bda90f1, tags={ info.nightscout.androidaps.receivers.KeepAliveReceiver$KeepAliveWorker } ]
2021-12-17 22:05:24.333 20623-20760/info.nightscout.androidaps E/AndroidRuntime: FATAL EXCEPTION: Thread-10
    Process: info.nightscout.androidaps, PID: 20623
    java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@ee6bd2f7: AdapterService getBondState
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2425)
        at android.os.Parcel.createException(Parcel.java:2409)
        at android.os.Parcel.readException(Parcel.java:2392)
        at android.os.Parcel.readException(Parcel.java:2334)
        at android.bluetooth.IBluetooth$Stub$Proxy.getBondState(IBluetooth.java:3236)
        at android.bluetooth.BluetoothDevice$3.recompute(BluetoothDevice.java:1595)
        at android.bluetooth.BluetoothDevice$3.recompute(BluetoothDevice.java:1590)
        at android.app.PropertyInvalidatedCache.query(PropertyInvalidatedCache.java:562)
        at android.bluetooth.BluetoothDevice.getBondState(BluetoothDevice.java:1632)
        at info.nightscout.androidaps.danars.services.BLEComm.connect(BLEComm.kt:109)
        at info.nightscout.androidaps.danars.services.DanaRSService.connect(DanaRSService.kt:101)
        at info.nightscout.androidaps.danars.DanaRSPlugin.connect(DanaRSPlugin.kt:148)
        at info.nightscout.androidaps.queue.QueueThread.run(QueueThread.kt:102)
        `

jroth-haj, 2021-12-17:
@tanja3981 did you try to clear all data from AAPS and also from bluetooth system app?
I also recommend a reboot after that.
Then AAPS should start with the new installation wizard.

tanja3981, 2021-12-17:
@jroth-haj I tried deleting data, caches,... Then I can start AAPS.
But as soon as I import my settings and restart (as requested), it crashes again.

MilosKozak, 2022-01-06:
https://github.com/nightscout/AndroidAPS/commit/445b1bb6ae8dedbef7d992e2c8f884620e69a540

closed 2022-01-06 by MilosKozak