nightscout/AndroidAPS #1074: 0 COB in calculator during Graph calculation

Philoul, 2021-12-15:
Annoying issue especially when I use Bolus reminder: I enter Carbs and 15 or 30 minutes later, when I have the reminder, I open Calculator to do my Bolus.
=> But when Alarm is raised, it's always just after a new BG is received. Then I have to wait sometimes more than one minutes to see correct COB in calculator. (My Samsung A40 is not very powerfull so it takes times to recalculate all graphs)
Sometimes I can see COB value in Overview, sometimes onlyI see n/a during countdown, but in both case, I have to wait the end of all graph calculation to have correct COB value in Calculator (otherwize calculation is done with 0 COB in Calculator so calculated bolus is wrong)

MilosKozak, 2021-12-30:
when history data is updated, for example smoothed BG values, all precalculated data from that time is invalidated and new calculation must be done. including COB. i don't see any way to change this

Philoul, 2021-12-31:
Is it possible to backup last calculated COB value before invalidate all calculated data and use this backup if a request is done during calculation?
The error using previous calculated COB value instead of last one (still not available) should be very small (compared to use 0 COB in wizard...)

MilosKozak, 2021-12-31:
well everything is possible but something is tough. maybe easier would be message in dialog "still waiting for COB calculation"

closed 2023-08-17 by MilosKozak