nightscout/AndroidAPS #1079: [3.0 Beta 11] OrangeLink not recognized

rustymonkey, 2021-12-16:
Upon launching 2.8.2.18-dev, no connection to the OL was possible. Clearing cache, restarting the phone and reseting the OL had no effect. Deleting the OL and initiating a scan didn't resolve the issue either because AAPS failed to find the OL (I've _never_ observed this before)!

This was previously working with all earlier builds (last tested was 2.8.2.16).

Tested with 2.8.2.18-dev = B11 / S7 running Android 11 / OL firmware 2.5 / MDT 554

[AndroidAPS_LOG_1639632763003.log.zip](https://github.com/nightscout/AndroidAPS/files/7724841/AndroidAPS_LOG_1639632763003.log.zip)

rustymonkey, 2021-12-16:
Note: reverting to 2.8.2.16 solved the problem confirming that it's a recent regression.

rustymonkey, 2021-12-16:
Further investigation reveals that this issue occurred due to a combination of:
1) #474 : this issue sometimes occurs on reboot, the workaround requiring deletion/addition of the RL/OL
2) A cosmetic bug in Beta 11 in which the RileyLink configuration dialog displays the scanned OrangeLink almost invisibly. In other words, when the user presses [Scan] on the RileyLink configuration panel, the RL is scanned, but displays so faintly that it's almost invisible! See below:
[image]

Philoul, 2022-01-02:
@rustymonkey can you test PR #1120 and tell me if it fix your issue (black text...) ?

closed 2022-01-03 by MilosKozak

rustymonkey, 2022-01-17:
Verified in 3.0 RC2. Thanks!

PR #1120 (2022-01-03): Add Style to fix black text
Try to fix #1079 (but I don't have any RL to test if it works or not)

@andyrozman can you tell me if #1079 is fixed or not ?