nightscout/AndroidAPS #1085: [3.0-beta9] NSClient - Incorrect vertical position of BG check on graph

vmichalec, 2021-12-17:
Hi, this is probably only displaying issue on NS client side, details in screenshots.

Here is bg check of 5.8 correctly displayed on aaps graph, also entry in treatments shows value corrrectly. (Sorry for non-english screenshots)
[image]

[image]

But on NS Client side the value is positioned on the bottom of the graph, not on correct y-axis (vertical) position.

[image]

And also in entries screen the value is not shown in the BG check entry.

[image]

vmichalec, 2021-12-18:
**[3.0-beta11]** I have checked it also on latest beta. NSClient side treatments entry now shows also the value - perfect. Now I can see also the real problem behind - in AAPS I entered blood glucose on action tab as value of 30.0 mmol/L but on NSClient side there is shown as 1.7 mmol/L so now it is obvious. Graph position of the entry on NSClient side is now lite bit higher so it looks like the graph is correct and shows the label on 1.7 y-axis position.

Moreover I'm using also 1.9.0.5 BYODA. Calibrations made here are shown correctly in AAPS and also NSClient.

Philoul, 2021-12-20:
30mmol for a BG value! Very high, but 1.7 is a wrong convertion mg/dl to mmol/l.
Did you see 1.7mmol in user entry or in another place?

vmichalec, 2021-12-21:
3.0-beta11 for aaps and nsclient, 14.2.5 nightscout version.

Doing again now: I entered BG Check in action tab of 30 mmol/L on aaps side, it becomes 1.7 mmol/L on nsclient side. Graph position of the entry on nsclient is slightly bellow 2 so I think it is only unit conversion problem.

Philoul, 2021-12-21:
Can you please send me a screenshot of "user entry" icon in both phones please to help me understand where is the bug

vmichalec, 2021-12-21:
AAPS:
[image]
[image]
[image]
[image]
[image]

NSClient:
[image]
[image]
[image]

Philoul, 2021-12-21:
@MilosKozak I confirm the bug, there is a wrong unit change somewhere between source phone and destination phone (see below send and received values in log)

Value sent by source phone is OK (Unit is MMOL recorded in database and sent in NSCLIENT)
```
10:53:43.752 [RxCachedThreadScheduler-95] D/DATABASE: [NSClientAddAckWorker.doWork$lambda-7():114]: Updated ns id of TherapyEvent TherapyEvent(id=4438, version=0, dateCreated=1640080403090, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=61c1a427cfbcaf0004001c7d, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1640080392000, utcOffset=3600000, duration=0, type=ANNOUNCEMENT, note=test notif, enteredBy=AndroidAPS, glucose=6.0, glucoseType=SENSOR, glucoseUnit=MMOL)
10:53:43.763 [pool-19-thread-2] I/NSCLIENT: [DataSyncSelectorImplementation.processChangedTherapyEventsCompat():536]: Loading TherapyEvents data Start: 4438 ID: 4438 HistoryID: TherapyEvent(id=4439, version=0, dateCreated=1640080403090, isValid=true, referenceId=4438, interfaceIDs_backing=null, timestamp=1640080392000, utcOffset=3600000, duration=0, type=ANNOUNCEMENT, note=test notif, enteredBy=AndroidAPS, glucose=6.0, glucoseType=SENSOR, glucoseUnit=MMOL) 
```

But value received on the other phone, unit is wrong (MMOL is replaced by MGDL)
```
10:53:48.987 [pool-18-thread-2] D/DATABASE: [NSClientAddUpdateWorker.doWork():291]: Inserted TherapyEvent TherapyEvent(id=1152, version=0, dateCreated=1640080428971, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=61c1a427cfbcaf0004001c7d, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1640080392000, utcOffset=3600000, duration=0, type=ANNOUNCEMENT, note=test notif, enteredBy=AndroidAPS, glucose=6.0, glucoseType=SENSOR, glucoseUnit=MGDL)
```

Philoul, 2021-12-21:
Additionnally record is correct in NS (units = mmol)
```
    {
      "eventType": "Announcement",
      "isValid": true,
      "created_at": "2021-12-21T09:53:12.000Z",
      "enteredBy": "AndroidAPS",
      "units": "mmol",
      "notes": "test notif",
      "glucose": 6,
      "glucoseType": "Sensor",
      "isAnnouncement": true,
      "identifier": "61c1a427cfbcaf0004001c7d",
      "srvModified": 1640080392000,
      "srvCreated": 1640080392000
    },
```

Philoul, 2021-12-21:
Very strange Breakpoint in `TherapyEventExtension.kt`line 72 (`therapyEventFromJson` function), and the json string looks like that:
"unit" = "md/dl" and I have a "mmol" key with value ???
```
{
    "_id":"61c2407fcfbcaf0004001dc4",
    "eventType":"Question",
    "isValid":true,
    "created_at":"2021-12-21T21:00:28.000Z",
    "enteredBy":"AndroidAPS",
    "units":"mg\/dl",
    "notes":"test",
    "glucose":6.5,
    "glucoseType":"Finger",
    "mills":1640120428000,
    "mmol":6.5,
    "targetTop":null,
    "targetBottom":null
}
```

It's not the json string got from NS (see below json strings with API V1 queries)
```
{
    "_id":"61c2407fcfbcaf0004001dc4",
    "eventType":"Question",
    "isValid":true,
    "created_at":"2021-12-21T21:00:28.000Z",
    "enteredBy":"AndroidAPS",
    "units":"mmol",
    "notes":"test",
    "glucose":6.5,
    "glucoseType":"Finger",
    "mills":1640120428000,
    "carbs":null,
    "insulin":null
},
```
Unfortunatly I can't find where json string is modified to make a clean fix...

MilosKozak, 2022-01-03:
it's NS bug. reported to NS team

closed 2022-01-03 by MilosKozak