nightscout/AndroidAPS #1086: Unexpected errors/retries during OmniPod DASH activation

vanelsberg, 2021-12-19:
AndroidAPS 2.8.2.18-dev
**Build:** e14cdd7c2
**Flavor:** Fullrelease
**Phone:** Samsung Note 9 (Android 10)

**Problem description**:
The phone never showed any problem activating DASH.
This time however I got 3 errors during activation. Activation continued successfully after retry on all errors. First one was about not seeing the Pod (needed 2 retries, second error complained about connection and service not found?)

**Additional information**
The log files show only 2 exception (I am sure I needed to hit the 'retry' button 3 times?

_09:32:31.387 [RxCachedThreadScheduler-88] E/PUMP: [DashInitializePodViewModel$doExecuteAction$1$1.invoke():66]: Error in Pod activation part 1
info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.FailedToConnectException: Failed to connect: 54:27:8D:44:1C:C6

09:32:43.335 [RxCachedThreadScheduler-82] E/PUMP: [DashInitializePodViewModel$doExecuteAction$1$1.invoke():66]: Error in Pod activation part 1
info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: Service not found: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f_

**Repro:** N.a.
Suspecting Pod or incidental BT problem causing this issue?
Will double check and update here for results on next activation.

**Log files:**
[AndroidAPS_LOG_1639903053512.log.zip](https://github.com/nightscout/AndroidAPS/files/7740652/AndroidAPS_LOG_1639903053512.log.zip)

vanelsberg, 2021-12-19:
Proposing to ignore this issue for now, unless more problems are reported.
Will monitor connection quality for this Pod and retest for a possible repro on next DASH activation.

robertrub, 2021-12-23:
Error while activating my last DASH. All went well, DASH on body and only _send profil and insert canula_ needs to be done. I hear the pod ticking, needle insertion and an error on AAPS. I hit _retry_ and it finished the activation process. 
Didn't have time to screen capture or get logs :(

robertrub, 2021-12-25:
[image]
I opened another issue. I'm not sure if it is the same thing or not #1096

avereha, 2021-12-29:
Fixed the issue reported by @robertrub with https://github.com/nightscout/AndroidAPS/pull/1108

vanelsberg, 2021-12-29:
> Will monitor connection quality for this Pod and retest for a possible repro on next DASH activation.
FWIW: Have not seen the problem I reported at last 3 Pod activations.

closed 2022-01-16 by MilosKozak