nightscout/AndroidAPS #1089: [Enhancement] Estimate remaining omnipod reservoir level using total insulin delivered?

legoandmars, 2021-12-20:
Since omnipod pumps don't report reservoir levels until they get below 50U - 1/4th of the actual reservoir capacity - glancing at the reservoir levels to gauge when a new pod is needed can be unhelpful a large majority of the time

What I *have* found very helpful is mentally subtracting 200U by the total insulin delivered, which gives me an estimate generally accurate within a few units

I think implementing something like this in the status light UI could be quite useful (displaying `~180U` instead of `50+U` if 20 units have been delivered, for example). However, not everybody fills their pod to 200U, which could complicate this too much to be worth adding

A couple of solutions I can think of are:
1) Displaying a range between `Math.max(minFillLevel - insulinDelivered, 50)` and `maxFillLevel - insulinDelivered` until the pump reports less than 50 units in reservoir - this might not fit in the status light screen without modification
2) Having the user input their fill level somewhere  - this might be too much of an unnecessary extra step just to get an estimate
             \* Could be locked behind an off-by-default setting for those (like myself) who would be willing to take the extra step

Would be happy to make a PR for this if it's deemed something add-able

robertrub, 2021-12-20:
Not everyone fills the pods with 200U. How can you tell what is left if you don't know how much was in the pod ?

vanelsberg, 2021-12-28:
> glancing at the reservoir levels to gauge when a new pod is needed can be unhelpful a large majority of the time

1. AAPS overview accurately displays the Pod age in days/hours. User can be expected to know Pod lifetime is 3 days.
2. The DASH tab shows the exact date and time when the Pod need to be changed ("Pod Expires")
3. Both the DASH tab and AAPS statistics accurately show information on insulin delivery.

Additionally to this Omnipod accurately reports the reservoir level when it gets below 50u. This is used to alert the user in case the Pod runs out of insulin before the "technical" end of the Pod.

I can not see the need for knowing the reservoir level (unless it gets near empty which is covered by current functionality).

vanelsberg, 2021-12-28:
@MilosKozak Proposing to close this issue as "no enhancement, covered by current functionality"?

MilosKozak, 2026-10-16:
V4 shows insulin delivered through pod so you can estimate remaining insulin above 50U when you know how much you filled

closed 2026-10-16 by MilosKozak