nightscout/AndroidAPS #1091: [Latest dev] No profile change if only the target area was changed

swissalpine, 2021-12-22:
If I change the target area in a Local Profile and save it, then trigger a profile change, the profile change is not performed and the previous targets remain active.
Workaround: Start a profile for a few minutes with a percentage change, then - when switching back - the changed Local Profile with the changed targets is active.
Version: Latest devs (already for a longer time)
Pump: Combo

Alamo04, 2021-12-29:
Confirmed
There are some fixes done in the past, but they didn't help in that case.

You must always have a small change in your basal profile (0,01) then the profile switch works without to change to a different profile first.

MilosKozak, 2021-12-30:
hmmm it works properly for RS/i

MilosKozak, 2021-12-30:
looking at te code it should work for Combo as well
Please provide logs

Alamo04, 2021-12-30:
Forced it for you @MilosKozak .
Changed only Target fom 100 and 90 to only 100.

Shown in treatment but nothing happens in AAPS.

[image]
[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7794204/AndroidAPS.log)

MilosKozak, 2021-12-30:
fixed by https://github.com/nightscout/AndroidAPS/commit/7243f8fb6645b8de9271f1a0b55352231b2c9015

closed 2021-12-30 by MilosKozak