nightscout/AndroidAPS #1095: [3.0-beta11] Seemingly random BS alerts

1401Ger, 2021-12-24:
I am getting bs measurement alerts with the following configuration:

Pump: Dana-i
CGM: Dexcom G6
AAPS: 3.0-beta11
Phone: Galaxy S10e, Android 11

The G6 sensor is connected to the latest BYODA app (G6 International v1.9.0.5) that is configured to send readings to AAPS. Everything is working fine most of the time, but both during the day and during the night, seemingly random alerts ("Alarm BZ-Messung", English: alert BS measurement) occur.

In the attached log this happend at
01:22:57.958
03:13:09.986
04:47:59.457

[image]

Both BYODA and AAPS have full permissions in the phone settings. Google Play Protect is deactivated. Energy Optimization is deactivated.

[AndroidAPS_LOG_1640328854736.log.zip](https://github.com/nightscout/AndroidAPS/files/7774520/AndroidAPS_LOG_1640328854736.log.zip)

1401Ger, 2021-12-24:
Update: This might be an issue solely related to BYODA, I'll try some changes and phone settings and will report back tomorrow.

MilosKozak, 2021-12-26:
this is message comming from pump

1401Ger, 2021-12-26:
> this is message comming from pump

Is there a way to figure out what caused this message or how to prevent it? So far I can't find any corresponding errors in the error log of the dana-i. If I can find out which alert or setting causes the message I can look for the appropriate setting.
I am a bit confused since the alert in AAPS refers to a BS measurement yet the G6 sensor is only connected to BYODA, not to the dana-i.

1401Ger, 2021-12-27:
Update: The alert seems to stem from a "Check BG" alert from the dana-i. According to the manual this alert is supposed to remind the user of checking the BG level after a bolus (for manual pump therapy). 
By setting alert melody to "0" in the doctors menu, these alerts are disabled. I suspect the alerts were triggered by SMBs.
Yet, I still have no idea why the alerts seem to occur at seemingly random instances of administed boli - it doesn't seem to correlate with the insulin volume of the SMB right before. Sometimes many SMBs were administerd for hours without this alert occuring in AAPS, and sometimes it occured sevel times per hour. 

If this is expected behavior, the issue can probably be closed. But maybe it is worth describing it in the documentation since I can't see any use in it for AAPS users.

closed 2021-12-27 by MilosKozak