nightscout/AndroidAPS #1098: [3.0-beta11] NSClient Synchronization

vmichalec, 2021-12-26:
Preferences -> Overview -> Advanced settings -> "Deliver this part of bolus wizard result [%]" is not transferred to nsclient and therefore calculator on nsclient side shows different results when this value is not 100% or not synced manually.

Philoul, 2021-12-28:
I'm not sure to understand your request or issue.
Settings must be done manually on each version of AAPS (or NSClient version for remote monitoring)
=> You have to be consistent when you modify settings on one side if you want to have same results on the other side.
Do yoiu want to automatically transfert some settings of AAPS to NSClient version ?

vmichalec, 2021-12-28:
Yes, that's exactly what I meant. Auto sync of this value would be nice. But of course, it's like minor enhancement.

(I think there is some synchronization mechanism already (like for min_5m_carbimpact), so maybe this field can be added to sync)

MilosKozak, 2021-12-30:
https://github.com/nightscout/AndroidAPS/commit/80e65824a3939bafcb1986f1cfa737a5608a332a

closed 2021-12-30 by MilosKozak