nightscout/AndroidAPS #1102: [3.0-beta11] Ruffy driver disappears (DeadObjectException)

ideaweb, 2021-12-27:
I already checked https://github.com/nightscout/AndroidAPS/issues/705 and I am not sure to re-open a new issue with the same problem, but I am in worry that the the closed one does not get any attention anymore when I add a comment.  If its wrong, sorry about it.

The hints offered by Leon does not help with my Jelly 2 smartphone (Android 11). I already compiled AndroidAPS with different SDKs. I get always DeadObjectException once a day. 

```
00:07:56.686 [main] D/CORE: [KeepAliveReceiver.onReceive():46]: KeepAlive received
00:07:56.765 [pool-15-thread-2] D/CORE: [KeepAliveReceiver$KeepAliveWorker.checkPump():143]: Last connection: 24.12.21 23:14
00:07:56.769 [pool-15-thread-2] D/PUMPQUEUE: [CommandQueueImplementation.add():147]: Adding: CommandReadStatus - READSTATUS KeepAlive. Status veraltet.
00:07:56.771 [pool-15-thread-2] D/PUMPQUEUE: [CommandQueueImplementation.notifyAboutNewCommand():182]: Starting new thread
00:07:56.774 [Thread-11306] D/PUMPQUEUE: [QueueThread.run():109]: connection time 0s
00:07:56.775 [Thread-11306] D/PUMPQUEUE: [QueueThread.run():115]: performing READSTATUS KeepAlive. Status veraltet.
00:07:56.776 [Thread-11306] D/PUMP: [ComboPlugin.getPumpStatus():349]: getPumpStatus called
00:07:56.777 [Thread-11306] D/info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter: [RuffyScripter.runCommand():233]: Attempting to run cmd: ReadPumpStateCommand{}
00:07:56.779 [Thread-11306] W/info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter: Disconnect after connect failure failed
android.os.DeadObjectException: null
	at android.os.BinderProxy.transactNative(Native Method) ~[na:0.0]
	at android.os.BinderProxy.transact(BinderProxy.java:540) ~[na:0.0]
	at org.monkey.d.ruffy.ruffy.driver.IRuffyService$Stub$Proxy.doRTDisconnect(IRuffyService.java:211) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.ensureConnected(RuffyScripter.java:477) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.runCommand(RuffyScripter.java:247) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.readPumpState(RuffyScripter.java:208) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.runOnConnectChecks(ComboPlugin.java:930) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.runCommand(ComboPlugin.java:872) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.getPumpStatus(ComboPlugin.java:354) ~[na:0.0]
	at info.nightscout.androidaps.queue.commands.CommandReadStatus.execute(CommandReadStatus.kt:23) ~[na:0.0]
	at info.nightscout.androidaps.queue.QueueThread.run(QueueThread.kt:118) ~[na:0.0]
00:07:56.781 [Thread-11306] E/info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter: CommandException while executing command
info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.commands.CommandException: Unexpected exception while initiating/restoring pump connection
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.ensureConnected(RuffyScripter.java:481) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.runCommand(RuffyScripter.java:247) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.readPumpState(RuffyScripter.java:208) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.runOnConnectChecks(ComboPlugin.java:930) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.runCommand(ComboPlugin.java:872) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.getPumpStatus(ComboPlugin.java:354) ~[na:0.0]
	at info.nightscout.androidaps.queue.commands.CommandReadStatus.execute(CommandReadStatus.kt:23) ~[na:0.0]
	at info.nightscout.androidaps.queue.QueueThread.run(QueueThread.kt:118) ~[na:0.0]
Caused by: android.os.DeadObjectException: null
	at android.os.BinderProxy.transactNative(Native Method) ~[na:0.0]
	at android.os.BinderProxy.transact(BinderProxy.java:540) ~[na:0.0]
	at org.monkey.d.ruffy.ruffy.driver.IRuffyService$Stub$Proxy.doRTConnect(IRuffyService.java:191) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.ensureConnected(RuffyScripter.java:457) ~[na:0.0]
	... 7 common frames omitted
00:07:56.783 [Thread-11306] D/PUMPQUEUE: [CommandReadStatus.execute():25]: CommandReadStatus executed. Reason: KeepAlive. Status veraltet.
00:07:56.863 [pool-15-thread-2] D/CORE: [KeepAliveReceiver$KeepAliveWorker.workerDbStatus():108]: WorkManager size is 868
00:07:56.884 [Thread-11306] D/PUMPQUEUE: [QueueThread.run():134]: queue empty. disconnect
00:07:56.894 [Thread-11306] D/PUMP: [ComboPlugin.disconnect():251]: Disconnect called with reason: Queue empty
00:07:56.895 [Thread-11306] D/info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter: [RuffyScripter.disconnect():197]: Disconnecting
00:07:56.897 [Thread-11306] D/PUMPQUEUE: [QueueThread.run():138]: disconnected
00:07:56.899 [Thread-11306] D/PUMPQUEUE: [QueueThread.run():149]: thread end
```

I am not an expert in coding Android/Java but found out (e.g. https://stackoverflow.com/questions/21826670/deadobjectexception-on-android-app):

"DeadObjectException - The object you are calling has died because its hosting process no longer exists."

I activated "Developer Options" and checked "Running Services" and as shown in screenshot below if everything is right, Ruffy driver is shown under AndroidAPS, but after DeadObjectException its gone. 

[image]

I have no idea it was killed by AndroidOS or it just crashes. I checked the configuration again and again, App blocker is disabled, battery saver and manager are off as well, Apps are whitelisted etc. but every day the driver disappears after some time, mostly at night.

```
adb shell dumpsys deviceidle
...
Whitelist user apps:
    net.dinglisch.android.taskerm
    org.monkey.d.ruffy.ruffy
    info.nightscout.androidaps
    com.android.phone
    com.joaomgcd.autonotification
    com.dexcom.g6.region3.mgdl
    com.android.bluetooth
```

And just adding the driver does not work:

```
adb shell dumpsys deviceidle whitelist +org.monkey.d.ruffy.ruffy.driver.Ruffy
Unknown package: org.monkey.d.ruffy.ruffy.driver.Ruffy
```

In the documentation I was reading that AndroidAPS is starting the driver (I hope I am not wrong), if reconnecting to driver fails and the process is gone, is it possible just restart the driver?

Thx in advance and thx for the great work!

swissalpine, 2021-12-27:
Same setup: Jelly 2, Ruffy, AAPS all latest devx -> no problems
Settings:
- App blocker on, time out: 10 min, no restrictions for AAPS und Ruffy
- Battery: AAPS and Ruffy not optimized

Alamo04, 2021-12-27:
Also on latest dev with Combo, Pixel 5, Ruffy and Dexcom Boyda.
No issues with my setup.

After first connection of my Combo with Ruffy and restart AAPS, I never saw this App or an issue with it again.

MilosKozak, 2021-12-29:
i'm afraid i cannot help in this case

closed 2021-12-29 by MilosKozak