nightscout/AndroidAPS #1105: [enhancement] - calculate and record carbs based on food inputs

erobinson, 2021-12-29:
Would like to be able to:

- Select a food from the FOOD tab and have it open the bolus wizard/calculator w/ the correct number of carbs
- Include fat & protein into carb calculations
- Record when specific foods were eaten (and then be able to see that in a history browser and BG levels after the meal)
- Have a meal calculator where multiple foods & portions can be added/adjusted and then the meal can record carbs or start the bolus wizrd/calculator
- nice to have - default the list of foods to foods frequently consumed at that time of day (i.e. default to show breakfast foods in the morning)

erobinson, 2021-12-29:
Initial concept:
[image]

swissalpine, 2021-12-31:
The ideas are great and the design looks good! I would use it!

pnharris02, 2025-01-22:
I like this idea of using the foods database to do bolus calculations.  I discussed something similar in one of the forums several months ago.  
Note that there is a relatively new ticket https://github.com/nightscout/AndroidAPS/issues/3761 that talks about entering carbs/100g directly from the bolus calculator.