nightscout/AndroidAPS #1116: APS 2.8.2.18 Spray distance Timestamp error in timer

Thomas-Wolff, 2022-01-01:
There is a periodic error in the timer when bolusing meals via the bolus suggestion calculator with a syringe interval. Carbohydrates were triggered with an interval of 10 minutes, in the timer these 10 minutes were not set but only 1 minute. In the APS main screen and treatment tab everything was booked and set correctly. Only affects the timer for the carbohydrate intake.

Enclosed are the corresponding screnshots and log file.

With the request to check, and best thanks Thomas

[image]

[image]

[image]

[image]

[AndroidAPS._2022-01-01_12-29-10_.10.zip](https://github.com/nightscout/AndroidAPS/files/7798058/AndroidAPS._2022-01-01_12-29-10_.10.zip)

AdrianLxM, 2022-01-02:
Thanks for reporting.
I guess the issue is with having a`Long` passed as time instead of an `Int`.
I'll PR a possible fix in the course of the day.

Thomas-Wolff, 2022-01-02:
Thanks for the information, I had already suspected something like that.

AdrianLxM, 2022-01-02:
To be fair, I didn't even get the timer to start on my phone with `Long` as datatype.
This PR fixes it for me: https://github.com/nightscout/AndroidAPS/pull/1119

@Thomas-Wolff would you mind testing if it solves the issue you are experiencing as well? Thanks!

Thomas-Wolff, 2022-01-02:
Hi Adrian, I have already installed the Build 31458fb2b. Today at supper I will test it and send an info.

Thank you very much and have a nice Sunday
Greetings Thomas

MilosKozak, 2022-01-02:
fixed by https://github.com/nightscout/AndroidAPS/commit/faca82f6e853dff5ab27425292dc542691d42245

closed 2022-01-02 by MilosKozak

Thomas-Wolff, 2022-01-02:
Milos I will update right away and test it at the next bolus.

Thanks for your effort, Greetings Thomas