nightscout/AndroidAPS #1117: DASH / EROS Show either percentage or units of bolus already injected

robertrub, 2022-01-02:
The watch face can tell me the percentage of the bolus injected but not the app :(

It would be helpful to show the percentage, or even better, the units, of bolus already injected on the phone interface too.

[image]
[image]

Thomas-Wolff, 2022-01-02:
I can also confirm this, would be very helpful as Robby describes it.

vanelsberg, 2022-01-02:
I think the progress bar gives a pretty good idea on progress. Can not see how adding percentage is this helpful.
_Question_: What can/should user do with this knowledge?

robertrub, 2022-01-02:
If you suddenly remark that you're going to get too much insulin and want to stop at a certain amount, you need the _U really injected_ information.

With the other pumps, you already have the precise amount injected, it is only missing in Omnipod pumps ;)

vanelsberg, 2022-01-02:
Ok.
Note user needs to know the number of units. Not the percentage which would only an calculated estimate from progress.

Showing the percentage would not be very efficient then?

1. User would need to note the exact percentage shown at the time bolus is stopped.
2. User would need to calculate XX% of of N units,
3. I think progress bar is based on estimated time, not on actual Pod ticks.
4. Executing a bolus cancel command takes some time so the last percentage shown may not be accurate?

I'd recommend users to **always** check treatments as it shows the most exact number of units for the bolus and may be more accurate.

(Btw: I think sole purpose of watch showing percentage is to show progress (as it does not have the progress bar). I would not trust using the last percentage shown for calculating the actual bolus amount.

robertrub, 2022-01-02:
That is why I said that knowing sent units is better tjan percentage.

I agree with you but even a _not very precise_ number will give a better precision than a blant progression bar. The injected U number might be a couple of tics wrong and each tic is only 0.05U... 

Of course, once you abort the injection, you should look up the real amount injected and add some more U if needed.

A progress bar is useless information. Either give an information that can be useful or just draw a dansing Gizmo :)

vanelsberg, 2022-01-03:
How about a 0...100 ruler under the progress bar (steps of 10%?) then?

robertrub, 2022-01-03:
Theo, why complicate things ? Watch face gives a percentage, same could be easily done on app UI too.

AAPS can even multiply the percentage with the total U to inject and give the amount already injected.

Why are you against more information ?

vanelsberg, 2022-01-03:
Against is the wrong word: I just would not vote **for** this as a feature as I can not see the value of it ;-)
And please note: this is just my personal view. Anyone is free to see this different.

vanelsberg, 2022-01-03:
Hi @robertrub,
Reading back I see myself circling around the basics of why I would not choose to show any percentage. I'll try to rephrase what I was actually trying to say as I see some danger in suggesting a percentage of the bolus being given.
(Please ignore my previous comments.... Sorry for the confusion?)

The progress bar is a generic way of AAPS to show progress based on the estimated total time needed to finish the task of administering a bolus. How this task is performed is up to the driver. It could very well include some pump initialization and finishing with the actual bolus being executed somewhere along the way (please correct me if I am wrong here?)

_In the context of the AAPS progress bar showing XX%, to the user this would suggest XX% of the bolus was administrated (your argument) which may be incorrect. For DASH the progress bar pretty well matches the actual bolus process, however last time I checked a DASH bolus of 1U finishes at about 90% in progress.
Have no idea how this is for other pumps._

Philoul, 2022-01-11:
In AAPS, when you have a total amount, and a percentage for a progress bar, it's easy to convert to approximate amount of injected units...
User Interface for Insight pump is a bit different...
[image]
if @avereha is ok it could be easy to use same UI for dash (and eros)

avereha, 2022-01-12:
Sounds good to me!

PR #1972 (2022-08-01): Update BOlus dialog with estimated amount of Unit injected
Use the same dialog than Insight for Dash and Eros Pumps
Fix #1117
closed 2022-07-25 by robertrub