nightscout/AndroidAPS #1118: [3.0-beta/2.8.2.18-dev] NSClient Profile display (window top) is not always updated according to the stored profile.

reijoahola-nightscout, 2022-01-02:
Tested with the latest 2.8.2.18-dev but the problem has been there also in earlier Beta and dev versions
AAPS app and NSClient has installed into the same phone (Samsung Galaxy S9 / Android 10)
AAPS / NSClent / Synchronisation settings : Upload Data to NS = ON, Receive Profile Store = ON, other settings OFF
Virtual Pump / Insight used in the test AAPS system

When Profile is cloned/stored/activated (without changes) in AAPS, it is normally stored into the pump (virtual pump) and shown
correctly in the top window display.
After a while the same is also updated automatically into NSClient Profile and into the top window display.

When after this, the profile is changed (DIA from 7.0 h  -> 9.0 h) it is updated into the top window display in AAPS but
it seems to be that it always fails to be updated into the top window display in NSClient. However it is updated into the profile in NSClient so the profile in NSClient is not in synch with the display.

When the prpofile is later edited again in AAPS (DIA 9.0 -> x.x) it may updated correctly also in NSClient but I am not sure does this happen always. So, there is something mysterious and random in these profile display updates.

MilosKozak, 2022-01-02:
i'm not sure i fully understand.
could you provide step by step instruction with screenshots how to reproduce?

reijoahola-nightscout, 2022-01-03:
I have to study this more why this problem sometimes happens and sometimes not. It may depend on the time difference between the first phase of profile change (a new profile cloned/activate) and the second phase (DIA changed in the the activated profile).

reijoahola-nightscout, 2022-01-07:
I simplified the test situation and tested now only AAPS app -> NSClient app synchronisation.
AAPS / NSClent / Synchronisation settings : Upload Data to NS = ON, Receive Profile Store = OFF, other settings OFF

If I clarify what I tried to test:
1 phase of test:
- Clone a new profile in AAPS app with a new name but parameters unchanged.
- Activate it an make Profile switch
- check that it is activated both in AAPS app and also NSClient app
The first phase of the test works always, no problems.

2nd phase of the test:
- Then do not wait long and open that profile again and edit DIA time (in my case 7.0h -> 9.0h)
- Activate the edited profile and make Profile switch
- Check that it is activated in AAPS app .....
- ... and then go to check it in NSClient app

It turns out that the profile is not updated NSClient app if you started 2nd test phase soon after the first phase.
If I wait 1 .. 2 minutes then there is not this problem.

Note: When I say that the profile (DIA) was not updated I mean the profile list/window opened from the top of the main screen.

MilosKozak, 2022-01-07:
it's NS related issue. I created PR https://github.com/nightscout/cgm-remote-monitor/pull/7285 to improve behaviour of NS sync.
So far events must be at least 1 min off each other for proper sync

MilosKozak, 2022-05-27:
currently NS dev branch is needed for proper sync

closed 2022-05-27 by MilosKozak