nightscout/AndroidAPS #1123: [3.0-beta10, 3.0-beta11] Wear OS watch Complications are not working

reijoahola-nightscout, 2022-01-03:
From version 3.0-beta10 when trying to read BG, IOB etc. data from AAPS watchface into other watcface "Complications" fields (e.g. 3100 Crucial watchface), no data is shown and only "-" text is shown. AAPS complications worked in earlie beta-versions. Also when reading data from xDrip+ into Complications filelds works normally so the problem is not in the watch software or 3100 Crucial watchface.

reijoahola-nightscout, 2022-01-03:
When using app-full-release-3.0-beta11 and wear-full-release-3.0-beta8, Complaints work.
If wear is beta11 then there is also big problems to keep AAPS watchfaces open and those seem to close or crash immedfiately when selected.

reijoahola-nightscout, 2022-01-05:
This another issue #1130 https://github.com/nightscout/AndroidAPS/issues/1130 may have something to do with problem.

MilosKozak, 2022-01-07:
fixed by https://github.com/nightscout/AndroidAPS/commit/fd814baa601f954baf5d251a4959f744598fa791

closed 2022-01-07 by MilosKozak