nightscout/AndroidAPS #1126: PineTime support?

DOFfactory, 2022-01-04:
I wonder whether anybody considered developing AndroidAPS monitor for [PineTime](https://www.pine64.org/pinetime/). According to the description it can have 1 week battery life, and as a purpose-made device might be best for privacy-preserving solutions. I am not a developer but as a power-user I am happy to test.

MilosKozak, 2022-01-04:
it would basically mean write AAPS from scratch. It's not Android

closed 2022-01-04 by MilosKozak

DOFfactory, 2022-01-05:
Hold on a minute, I was under the impression that you usually pair the watch with a phone, and then you can share and visualize data on it: https://lupyuen.github.io/pinetime-rust-mynewt/articles/timesync

Since one can develop a Hello World app for the watch (https://mynewt.apache.org/latest/tutorials/blinky/pinetime.html), I was thinking it might be possible to write an app that visualizes basic info from AAPS. Obviously, AAPS will need to have some module built in it or something to provide the data. You say this is not possible? I am not asking rewriting AAPS, maybe a module for sharing basic data with a corresponding PineTime app.

MilosKozak, 2022-01-05:
if there is documentation it is possible.. But it's lot of work. There must be really large amount of users of this watch to make someone start work on this