nightscout/AndroidAPS #1130: [3.0-beta11] Wear watchfaces do not open

reijoahola-nightscout, 2022-01-05:
3.0-beta11 and dev version afterr it
TicWatch Pro

When trying to select some of AAPS watchfaces, they typically does not open but are cloasedf or crashed
and only grey display with time is shown.

In earlier beta versions (at least in 3.0-beta8) watchfaces work normally.

AdrianLxM, 2022-01-05:
Possible duplicate of https://github.com/nightscout/AndroidAPS/issues/1122

reijoahola-nightscout, 2022-01-05:
I started a new project from a clean table and compiled the latest dev version 2.8.2.18-dev.
Still the same problem when trying to select AAPS watchfaces.

Andries-Smit, 2022-01-05:
Do you have some log details? https://developer.android.com/training/wearables/get-started/debugging

reijoahola-nightscout, 2022-01-05:
Oops, this "Debug a Wear OS app" is something new for me. I have to study this.
I will send logs later if I can achieve something useful. May be tomorrow I have time to concentrate on this

Is it so that others cannot reproduce this, I thought that this must be very common problem because
watchfaces work when I install the older version into the watch (wear-full-release-3.0-beta8) so it should
indicate that my watch is not broken or something like that.

reijoahola-nightscout, 2022-01-06:
Not yet successful with "Debug a Wear OS app" but I found one clue about possible reason.
I had one watchface (3100 Crucial) that had selected "Complaint" from AAPS COB and as it is not working (Issue #1123) it showed only -. When I turned off this Complaint selection then all AAPS watchfaces started to work. Probably Complaints-bug somehow disturbs also AAPS watchfaces if some watchface/Complaint try to read AAPS Complaints.

reijoahola-nightscout, 2022-01-06:
[wear-2.8.2.18-dev_2022_01_06_log.txt](https://github.com/nightscout/AndroidAPS/files/7821780/wear-2.8.2.18-dev_2022_01_06_log.txt)
Here is log from a situation (around time 14:20 ... 14:21) where I at first enabled again "3100 Crucial" watchface Compliant from AAPS IOB. Then selected AAPS watchface "Digital Style" and it was not successful and gave grey screen with time. Then I again disabled "3100 Crucial" watchface problem Compliant and could after some watchface selection again have AAPS "Digital Stye" watchface. I hope this logreport helps to solve the problem.

MilosKozak, 2022-01-07:
fixed by https://github.com/nightscout/AndroidAPS/commit/fd814baa601f954baf5d251a4959f744598fa791

closed 2022-01-07 by MilosKozak