nightscout/AndroidAPS #1132: [BETA 11] Problem with data processing

69nobody, 2022-01-06:
New data such as delivered boluses or eaten carbohydrates cannot be processed and are not displayed.
Unpairing and re-pairing unfortunately only worked for a short time.
Then the same was displayed as marked in red in the picture. "...TempStop 06:24..."
In the statistics Dana-i tab is displayed under error, event is processed TEMP_STOP 06:24.

This is really weird, right after delivering a bolus comes "loading events" then the already known red framed is displayed again.

Bolus is delivered, the number of units does not change in the IOB display. The same applies to COB.

Translated with www.DeepL.com/Translator (free version)

69nobody, 2022-01-06:
I was able to solve my problem by resetting the databases.
I then imported a copy of my settings and activated my profile.
IOB and COB are updated and displayed again.
The TEMPSTOP entry is no longer displayed.

MilosKozak, 2022-01-06:
https://github.com/nightscout/AndroidAPS/commit/3a30f73aa11273e7e44938be60022ba9c668ea50

closed 2022-01-06 by MilosKozak