nightscout/AndroidAPS #1136: UI recalculates a lot on received NS confirmations

tanja3981, 2022-01-06:
AAPS: 3.0beta11
NS: 14.2.6
Android: 12

I paused NSClient for a while. When started, the queue was delivered as it should.
However, the AAPS main screen kept showing the recalculating message, starting again and again and 400, finally Android asked me to stop the app multiple times.

Video:
https://drive.google.com/file/d/123NaM3Wwh62HejVCtPNQUP6JdMAOgUN_/view?usp=sharing

tanja3981, 2022-01-06:
Logfile: [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7824205/AndroidAPS.log)

catherinefranq, 2022-01-09:
probleme avec BOYDA qui a 23mn de retard dans APS j'ai dus repassé a NS pour avoir mes glycemies à jour cela fait 2 ou 3 jrs que BOYDA rame sur APS je cours avec APS 3.0beta11 nightsout version 14.2.2 et BOYDA 1.9.0
si on peut m'expliquer pourquoi cela fonctionnais bien avant ?
je précise je n'ai pas supprime le pont dexcom dans nightsout

tanja3981, 2022-01-09:
@catherinefranq This has nothing with the bug reported here. Please open an own issue for it.

MilosKozak, 2022-01-30:
https://github.com/nightscout/AndroidAPS/commit/8c28dd0f527f246fc613978b72cfb1a11e49446e this should help a lot

closed 2022-02-04 by tanja3981