nightscout/AndroidAPS #1137: [3.0-beta11] NSClient Synchronisation setting has Receive APS offline events. Is it needed ?

reijoahola-nightscout, 2022-01-07:
When testing synchronisation of various things between AAPS and NSClient I found that there is a setting "Receive APS offline events" in NSClient app. It specifies that "Accept APS offline events entered through NS or NSClient". However I cannot find any way of giving "APS offline events" (Suspend Loop, Disconnect Loop etc.) from NSClient. Have I misunderstood something or is this setting unnecessary ?

MilosKozak, 2022-01-07:
you'll see in nsclient timerange where aaps was off .... that's all

closed 2022-01-07 by MilosKozak

reijoahola-nightscout, 2022-01-07:
Yes, I noticed that timerange can be seen also in NSClient app when the operation is started from AAPS app. But the question was is AAPS synchronisation setting needed because the operation cannot be started from NSClient.

MilosKozak, 2022-01-07:
no at the moment but maybe in the future

reijoahola-nightscout, 2022-01-07:
> no at the moment but maybe in the future

Should the setting be grayed as long as it has no real function.