nightscout/AndroidAPS #1139: Dev 19 (and before) / Too many alert beeps

robertrub, 2022-01-09:
Not urgent. This happened several times with the older dev versions too. 

AAPS sends a notification via Android and it "beeps". In some occasions (can't find out when :(  ) it beeps 3-5 times. I copied the logs in case you can see the reason in them.  Look at time 11:46

[AndroidAPS_LOG_1641638856393.log.zip](https://github.com/nightscout/AndroidAPS/files/7834876/AndroidAPS_LOG_1641638856393.log.zip)

[image]

MilosKozak, 2022-01-09:
i see notifications only from automations

robertrub, 2022-01-09:
Tnx. I have automations to raise target when there is a hypo risk. Maybe this automation and "x grams of carbs needed" alert is the reason for the multi beeps.

I'll keep watching and report more or close the issue accordingly.

closed 2022-01-16 by MilosKozak