nightscout/AndroidAPS #1141: [3.0beta11] NSClient does not see temp target

kristinsmotz, 2022-01-09:
Just discovered that when a temp target is set either manually or via automation on AAPS phone, it can not be seen on NSClient phone.

Aaps:
[image]
NSClient:
[image]

[AndroidAPS_LOG_1641752708207.log.zip](https://github.com/nightscout/AndroidAPS/files/7835660/AndroidAPS_LOG_1641752708207.log.zip)

MilosKozak, 2022-01-10:
if the problem is on NS side I need logs from NS as well. and time of occurence

kristinsmotz, 2022-01-10:
I will try to recreate this afternoon, and include logs from both phones (AAPS phone currently at school). I checked in Nightscout reporter (https://nightscout-reporter.zreptil.de/) and was able to find all the temp targets from yesterday in a report there, so info is coming through to NS, but possibly not to NSClient. 
[image]

As per right now I checked NSClient - Treatments - temp target tab, and could not find any temp targets there for yesterday except one that was cancelled due to no connection I believe. 
[image]

kristinsmotz, 2022-01-10:
I'll try tonight to reproduce and export logs from both phones, then update to RC-1 and try to reproduce error there as well.

kristinsmotz, 2022-01-10:
I just remembered; i updated our nightscout site 5th january at night, in order to fix a previously reported error with BG check having wrong value in NSClient. I did an update to the latest dev on nightscout. So I would guess that something in that build is causing this error for me now. If updating to RC-1 tonight does not help I will roll back NS to latest master, and see if that does it then report the error there as well.

kristinsmotz, 2022-01-10:
I decided to revert to master on my nightscout site and can now see temp targets in NSClient. I suspect this commit on the nightscout dev is what caused this issue: https://github.com/nightscout/cgm-remote-monitor/commit/aca8d5ab492f1e356c36f340102044dc961d82aa#diff-5a16ec3fb169ae3b2b0b4d0a810080d451c8910345d86280d1cc2c67d4a178a0

MilosKozak, 2022-01-12:
so passing TT to NSClient is not working with current NS dev?

kristinsmotz, 2022-01-12:
That is correct, or at least with the dev as it was on 5th January.

kristinsmotz, 2022-01-13:
Today had another occurrence of TT not showing up in NSClient, only on AAPS phone. We updated to RC-1 a couple of days ago. The TT came from an automation that runs every Thursday at 10.00. it also posts a message to graph area "Fysak torsdag 10.30-12.00", and sets a temporary 80% profile, both of which showed up on both phones.
 
Screenshots from AAPS: 
[image]
[image]
[image]

Screenshots from NSClient:
[image]
[image]

Log from AAPS: 
[AndroidAPS_LOG_1642089548988.log.zip](https://github.com/nightscout/AndroidAPS/files/7864355/AndroidAPS_LOG_1642089548988.log.zip)

 
If it will help you investigate further, our NS site is https://heddamamonitor.herokuapp.com/
This time it appears that NS didn't catch the TT either:
[NS reporter log.pdf](https://github.com/nightscout/AndroidAPS/files/7864748/NS.reporter.log.pdf)

[image]

I'm having trouble with logs in NSClient, they're either only being put in the /logs folder at random times, or being deleted in random order. I'd really appreciate any advice to fix this so I always can have the logs for the last 24 hours available. Currently this is what the folder on my phone contains:
new logs every 10-20 minutes since 13.26, but still containing logs from 1.41, 2.01 and 2.16 this morning.
[image]

MilosKozak, 2022-01-16:
whithout NSC logs is not possible to resolve the issue. having so many files means there is too much logging enabled. give me a sample of NSC log

kristinsmotz, 2022-01-16:
[AndroidAPS_LOG_1642369862613.log.zip](https://github.com/nightscout/AndroidAPS/files/7878042/AndroidAPS_LOG_1642369862613.log.zip)
[AndroidAPS_LOG_1642369878670.log.zip](https://github.com/nightscout/AndroidAPS/files/7878043/AndroidAPS_LOG_1642369878670.log.zip)
[image]
[image]
[image]

I just did a set of tests to reproduce. 

First test: manual TT , showed up in NSClient
Second test: automation setting TT and Temp profile, did not show in NSClient
Third test: automation setting TT and note, showed up in NSClient. 

Added screenshots from treatments to easily see times to look for in log.

MilosKozak, 2022-01-22:
sorry i need both: screenshot + logs from AAPS and the same from NSC. I'm not sure what is what and second logs doesn't cover 22:15

kristinsmotz, 2022-01-24:
I understand. Having trouble with getting phones to send correct logs when I try. I will try to review log settings on both phones and recreate the errors. Do you have recommendations for what to log which will give you enough info without flooding the log with irrelevant stuff? Would this be sufficient on the NSClient phone?

[image]

MilosKozak, 2022-01-24:
no need to enable anything special.  just logs +- 1h from the event from both devices

kristinsmotz, 2022-01-25:
OK here goes. Started a TT from Automation at 21.00 (With a temp profile included in the same Automation). 
Info from NSClient phone:
NSClient 3.0-rc3
NSCLIENT
Build: 617cdd681-2022.01.17-16:02
Remote: https://github.com/nightscout/AndroidAPS
Flavor: nsclientrelease
Nightscout version: 14.2.5
Log: 
[AndroidAPS_LOG_1643142836954.log.zip](https://github.com/nightscout/AndroidAPS/files/7937130/AndroidAPS_LOG_1643142836954.log.zip)
Screenshots of Overview and TT tab in Treatments on NSClient phone:
[image]
[image]

From AAPS phone:
AndroidAPS 3.0-rc3
Build: 617cdd681-2022.01.17-15:51
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.5
Engineering mode enabled
Logs:
[AndroidAPS_LOG_1643142902100.log.zip](https://github.com/nightscout/AndroidAPS/files/7937145/AndroidAPS_LOG_1643142902100.log.zip)
Screenshots from Overview and TT tab in Treatments:
[image]
[image]

kristinsmotz, 2022-01-25:
As previously mentioned, it appears this error only occurs when using an Automation to set a temp % profile AND a temp target in the same automation.

MilosKozak, 2022-01-26:
related to bug in NS
I already created a fix for that https://github.com/nightscout/cgm-remote-monitor/pull/7285 but it's not part of NS yet

closed 2022-01-26 by MilosKozak