nightscout/AndroidAPS #1145: 3.0beta11

catherinefranq, 2022-01-10:
probleme avec BOYDA qui a 23mn de retard dans APS j’ai dus repassé a NS pour avoir mes glycemies à jour cela fait 2 ou 3 jrs que BOYDA rame sur APS je cours avec
APS 3.0beta11 nightsout version 14.2.2 et BOYDA 1.9.0 si on peut m’expliquer pourquoi cela fonctionnais bien avant ?
je précise je n’ai pas supprime le pont dexcom dans nightsout.
finalement je suis passé avec nightsout cela ma créer pleins de doublon impossible a supprimer de ce fait la boucle ne fonctionnait pas comme il faut j'ai décider de désinstaller l'appli et fermer mon tel et réinstaller l'appli et remettre mes préférences j'ai eu une hyper de 22h00 à 3h00 ensuite sa c'est remis bien. Donc encore quelques beug qui sont gênant!

Philoul, 2022-01-11:
It's better writing in english here (we are very few fluent in french here...)
It seems you have settings in your config (dexcom bridge should be disable, data upload to NS enable from AAPS, all data download from NS disable to avoid duplicate entries).
I sometimes have some troubles with data arriving late from boyda (between 10 and 15/20min), but it's always when I use some specific other apps on the same phone (some games...). And when I close these specifics apps, byoda and aaps resynchronise data slowly (5min by 5 min) within 1h/2h...
Do you use other apps on your phone?
If your prefer french, you can PM me on discord...

Philoul, 2022-01-12:
I had to take my car this morning and I had same problem.
I started waze in frontend at ~8:30 (aaps running in background and I check my bg with my watch when I'm driving). I already did that hundred of times in the past...
But at 8:50 I noticed in my watch that my bg value was outdated, I immediatly put aaps in frontend, last received bg 20 min ago (while bg received in byoda was up to date).
During ~45min I still received every 5 min an outdated bg value that was sent by G6 sensor ~16/17min ago
I restarted AAPS at 9:34, same problem.
At ~9:50 I reboot my phone and restarted everything.
At 10:00 (10 min after restart, so an outdated bg that was received 31 min ago) was replaced by a 5min ago bg value (loop could restart 1h30 after last run), and during 45 additional minutes, I received every 5 min a bg value reveived 6 or 7 min earlier.
It's only at 10:45 min that everything was correct again with an up to date bg value (sent ~1min ago).
During all these 2h15min, byoda was always up to date, it's only aaps that was 3 or 4 values late compare to byoda.
I'm sorry to be a bit long in my explains, but I don't have my computer to analyse logs to help analysis...

I just attached log files before removal (I hope it's correct one... tell me if you need earlier or later)

[AndroidAPS._2022-01-12_09-48-40_.0.zip](https://github.com/nightscout/AndroidAPS/files/7854872/AndroidAPS._2022-01-12_09-48-40_.0.zip)
[AndroidAPS._2022-01-12_09-34-25_.0.zip](https://github.com/nightscout/AndroidAPS/files/7854873/AndroidAPS._2022-01-12_09-34-25_.0.zip)

Note: byoda 1.7.0.3 ok since install 1 year ago
AAPS 2.8.2.19-dev built 2 days ago

Philoul, 2022-01-12:
2 screenshots made (one before restart, 2nd after) one manual bolus made at 9:38 after verification in byoda that my bg was still rising...
[image]
[image]

catherinefranq, 2022-01-13:
bonjour
j'ai supprimé le bridge dans heroku et pour l'instant j'ai plus de problèmes à voir dans le temps merci

Philoul, 2022-01-13:
@catherinefranq Peux-tu renommer l'issue stp ?(bouton Edit à côté du nom), car "3.0beta11" ne décrit pas le problème, je te propose de mettre "BG values received in AAPS more than10min after available in BYODA", je ne peux pas la renommer de mon côté...
@MilosKozak I asked Catherine to rename this issue...

Philoul, 2022-01-13:
I started to analyse log file (line filtered with `[DexcomPlugin$DexcomWorker.doWork():154]: Inserted bg GlucoseValue`
I compared "DateCreated" and "timestamp" on each line and between each 2 lines (see Excel file)
- exactly 5 min (+/- 2s) between 2 timestamp => OK
- less than 1 min between date created and timestamp when everything is Ok (before 8:00 and after 10:46)
- duration between date created and timestamp increase slowly (but remain < 5 min between 8:00 and 8:17)
- I then received new data only 8/10 min after previous one (after that at 8:47 I had 16 min between Date Created and Timestamp)
- It's only 10 min after restart that I received within 1s five BG values

[Byoda_Analysis.xlsx](https://github.com/nightscout/AndroidAPS/files/7866375/Byoda_Analysis.xlsx)

I hope this will help analysis...

MilosKozak, 2022-01-14:
I'm afraid I cannot help. AAPS receives broadcast later. Impossible to say if it was broadcasted late or delayed.
I believe the issue is related to phone

closed 2022-01-16 by MilosKozak

Philoul, 2022-01-16:
I think Room has increased the CPU workload, and we have this issue among several other side effects (popup "AndroidAPS not responding" on main graph rescale or higher consumption of the phone)
If we can reduce the amount of calculations done when receiving new BG, I think (hope) we should improve the situation.

MilosKozak, 2022-01-16:
calculations would not delay broadcasts by minutes