nightscout/AndroidAPS #1150: [3.0-rc1] Some strange icons on bolus wizard

vmichalec, 2022-01-10:
I noticed some strange icons on bolus wizard in packed form

[image]

vs. unpacked
[image]

parapenT1sta, 2022-01-10:
Are you serious? What is strange?
It means to show you "BG" "Trend" "COB" and "IOB" options... that when you click on view they disappear because they are shown on the list...

vmichalec, 2022-01-10:
Ok, so it is a feature. Thanks.

closed 2022-01-10 by vmichalec

parapenT1sta, 2022-01-10:
> Ok, so it is a feature. Thanks.
[image]