nightscout/AndroidAPS #1151: Dana I Beta 11 stuck in pairing Loop

pbloom209, 2022-01-11:
Changing phone to Pixel, 
Restore my settings from preferences.
Started to pair pump and I entered the wrong PIN.

Now it is stuck in a loop.

I have cleared the Bluetooth stacks.

restarted phone.

The pump can be paired to Anydana.  Which I have now removed but it cannot pair to AAPS.  See video

https://user-images.githubusercontent.com/37948551/148898789-3f8217d5-5167-4775-8ef5-34fdc80ef0ea.mp4

.

MilosKozak, 2022-01-11:
make sure you removed pump from paired devices in android
long click on BT icon on Dana tab can reset pairing info inside AAPS

pbloom209, 2022-01-11:
I did all that didn't work. So I had to revert to the troublesome Samsung. Next time I will factory reset the pixel

pbloom209, 2022-01-12:
I am trying again with the Pixel, Android 12, still same issue.  I have done the clearing of pair data inside AAPS, and the pump is not paired to the phone.  Any other suggestions

MilosKozak, 2022-01-12:
are you able to pair pump with clear AAPS?

pbloom209, 2022-01-12:
No it didn't work.

I changed the settings in developer to 1.3 for AVRCP. it finally connected then I went to pair to the car and the whole phone reset and crashed.

Using a pixel 5 Android 12

MilosKozak, 2022-01-12:
if it doesn't work on factory reseted phone and fresh installed AAPS then I'm afraid I cannot help

closed 2022-01-12 by MilosKozak