nightscout/AndroidAPS #1152: [3.0beta11] Dash temporary basal rates precision: calculating plus of every 0,01 instead of 0,05

filipwoz, 2022-01-11:
Dash pump insulin dosing precision is 0,05 IE. When an over 100% temporary Basal rate is running the sum of the "plus" insulin is being counted with the precision of 0,01 instead of 0,05. I think the "plus" insulin should be added only in 0,05 steps.

avereha, 2022-01-17:
Could you please share where do you see this information(maybe with a screenshot)?

filipwoz, 2022-01-19:
I mean a situation like here:
[image]
- A temporary higher basal has started at 17:31
- It counts the IOB up in 0,01 steps although Omnipod has the precision of 0,05
Might be also that I am wrongly thinking:
- IOB is not the same as given insulin and the counter in the screenshot is counting both up and down at the same time (up for given insulin and down for already used insulin) so it is anyway not easy to count the expected result. But I have never something like a step going up by 0.05 thus my issue
- Other wrong thinking from my side might be: it might be much more difficult to find out **when** exactly the additional 0,05 injection steps have happened in the Pod thus as a good estimation it is better to count continuously by 0,01 and on average you will be quite close to truth (under the motto: in Diabetes is nothing precise and it is always an estimation) So in this case KISS and YAGNI principles would suggest to leave it as it is.

avereha, 2022-01-21:
I don't think this is a real bug. IOB can have lower resolution than the minimum pod delivery step.

MilosKozak, 2022-01-22:
@avereha is right

closed 2022-01-22 by MilosKozak