nightscout/AndroidAPS #1153: [3.0 RC1] profile shift not resulting in Bolus Wizard

hartlijn, 2022-01-11:
Shifting profile does not lead to correct ratios to be used in Bolus Wizard.
I shifted - 1 hour and in the wizard still the Carbratio of the original unshifted value is used. It should have used 5.5 but it took 6.2. 
[image]
[image]
[image]
[AndroidAPS_LOG_1641903106869.log.zip](https://github.com/nightscout/AndroidAPS/files/7846460/AndroidAPS_LOG_1641903106869.log.zip)

Philoul, 2022-01-11:
I don't think there is a bug here...
In your example, you shifted your profile 1h earlier, so at 1.06 PM, AAPS will take values of 2.06PM in your profile, and IC is 6.2 in your profile between 1PM and 5PM...

hartlijn, 2022-01-11:
Hmm, how stupid. Sorry.

closed 2022-01-11 by hartlijn