nightscout/AndroidAPS #1158: [3.0 RC1] VersionChecker - logs on the first day that the new RC1 version is expired

filipwoz, 2022-01-12:
Yesterday I have installed the 3.0 RC1 and in the evening I got a notification:

Notification received: Neue Version seit mindestens 1 Tagen verfügbar! Rückfall zur sensorunterstützten Pumpentherapie nach 7 Tagen, Loop wird nach 14 Tagen deaktiviert.

Looks like some bug in the version checker handing:  3.0 RC1 was published yesterday and the notification says it is already expired and some newer version is available since at least one day (but 3.0 RC1 is the newest one).

In log I have found:
21:04:03.542 [RxCachedThreadScheduler-9] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: Neue Version seit mindestens 1 Tagen verfügbar! Rückfall zur sensorunterstützten Pumpentherapie nach 7 Tagen, Loop wird nach 14 Tagen deaktiviert.
21:04:03.546 [RxCachedThreadScheduler-19] D/APS: [APSResult.isChangeRequested():310]: DEFAULT: Closed mode
21:04:03.551 [RxCachedThreadScheduler-19] D/APS: [APSResult.isChangeRequested():310]: DEFAULT: Closed mode
...
21:04:03.590 [Thread-2564] D/CORE: [VersionCheckerUtils.onOlderVersionDetected():113]: Version newer than master. Are you developer?
21:04:03.616 [Thread-2563] D/CORE: [VersionCheckerUtils.onOlderVersionDetected():113]: Version newer than master. Are you developer?
21:04:03.630 [Thread-2564] D/CORE: [VersionCheckerUtils.onExpiredVersionDetected():142]: Version 3.0-rc1 expired.
21:04:03.630 [Thread-2563] D/CORE: [VersionCheckerUtils.onExpiredVersionDetected():142]: Version 3.0-rc1 expired.
21:04:03.632 [RxCachedThreadScheduler-9] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: Version 3.0-rc1 läuft am 31.01.22 ab

mxsrm, 2022-01-12:
Can confirm this bug is still existing, even after the `versions` branch was updated.

winni67, 2022-01-13:
I can confirm this behavior.

MilosKozak, 2022-01-14:
it looks like this behavior is correct. RC really expires

closed 2022-01-14 by MilosKozak

filipwoz, 2022-01-14:
@MilosKozak: but as far as I understand it expires on 31.01.22 and the error message says:
**"Neue Version seit mindestens 1 Tagen verfügbar!"**
Behavior might be correct but the message was confusing also for other uses in the de.loopercommunity.org, as it suggests that a newer version than the RC1 already exists.

mxsrm, 2022-01-14:
Yes but rc1 is telling us there is a new version available for at least 1 day and it will fallback to LGS in 7 days.