nightscout/AndroidAPS #1159: [3.0 RC1] Dash pod initialization after discarding pod is failing on a wrong bluetooth address

filipwoz, 2022-01-12:
Repro Steps:
- try to deactivate dash pod
- break up the procedure and click on Discard Pod
- try to activate a new pod
Result:
Unhandled java Error message: trying to connect with one bluetooth adress but some other one is already available
Expected:
After discarding a pod a new pod should be possible to activate without issues

Seems that by discarding a Pod cleanup of the buetooth connection address does not happen.

I will provide the logs and the exact error message later today.

avereha, 2022-01-21:
How did you "break up the procedure"?

closed 2022-01-22 by MilosKozak

filipwoz, 2022-01-22:
> How did you "break up the procedure"?

I am not sure as my son did it without me looking at than. I thought you could read it from the logs.
I think he was telling that it was taking a long time (or even showing a timeout message) and then he pressed the button to discard the pod.