nightscout/AndroidAPS #1161: Connection to OL lost, unable to scan to reconnect.

Jane49, 2022-01-12:
Problem occurred from around 9.30a.m. and was resolved around 14:30 12th Jan 

Using Pixel 5 / Android 12 / Eros / OrangeLink v2.5 (not Pro) / Fossil Sport
AAPS 3.0-rc1 2022.01.09-21:30

No connection to RileyLink status.  Have had this with 2.8.2 but did manage to resolve it by changing pump driver to VL and then back to Omnipod - this may be related.

Unable to scan for OL.  Tried reflashing OL, reinstalling AAPS 3.0-rc1, removed all other BT devices, cleared phone BT storage and cache, multiple reboots etc.  and new batteries.  Still unable to scan for OL

Throughout all attempts Pod Management Screen displayed 
Errors : 'Rileylink address invalid' 
 
This was the same message in 2.8.2 which could eventually be resolved by changing the pump driver.

Workaround to resolve the problem:
Install 2.8.2 on top of 3.0 rebooted
wait .... then pump driver changed......  OL connected and it was ready (first attempt I became impatient, scanned and it got stuck again so installed 2.8.2 twice)
rebooted
installed 3.0 on top of 2.8.2
watched and waited waited a bit ........ connected 

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7855591/AndroidAPS.log)
[AndroidAPS._2022-01-12_06-15-34_.0.zip](https://github.com/nightscout/AndroidAPS/files/7855594/AndroidAPS._2022-01-12_06-15-34_.0.zip)
[AndroidAPS._2022-01-12_00-00-03_.0.zip](https://github.com/nightscout/AndroidAPS/files/7855598/AndroidAPS._2022-01-12_00-00-03_.0.zip)
[AndroidAPS._2022-01-12_13-11-14_.0.zip](https://github.com/nightscout/AndroidAPS/files/7855600/AndroidAPS._2022-01-12_13-11-14_.0.zip)
[image]

n0rbertg, 2022-01-12:
+1

The same happened to me as well. 
**Phone**: Pixel5 with Android 12 and OL Pro (but tried with other variants)
**Ver**: AndroidAPS 3.0-rc1 (tried with other - older - dev builds as well and was unable to scan so not sure when was this introduced

The issue is easily reproducible by removing the RL, no device shows up in SCAN. (I have tried with OL Pro, EmaLink and original RL as well, none shows up). Only workaround is to add the RL using an old release and switch back.

Log

```16:02:39.720 [main] D/PUMPBTCOMM: [RileyLinkBroadcastReceiver.onReceive():104]: Received Broadcast: AAPS.RileyLink.RileyLink_Disconnect
16:02:39.721 [main] W/PUMPBTCOMM: [RileyLinkBLE.disconnect():174]: Closing GATT connection
16:02:39.728 [main] I/PUMP: [RileyLinkServiceData.workWithServiceState():95]: RileyLink State Changed: BluetoothReady 
16:02:39.734 [Binder:9731_6] W/PUMPBTCOMM: [RileyLinkBLE$1.onConnectionStateChange():362]: onConnectionStateChange SUCCESS DISCONNECTED
16:02:39.734 [main] D/PUMPBTCOMM: [RileyLinkBroadcastReceiver.onReceive():104]: Received Broadcast: AAPS.RileyLink.RileyLink_Disconnected
16:02:39.735 [main] I/PUMP: [RileyLinkServiceData.workWithServiceState():95]: RileyLink State Changed: BluetoothError  - Error State: RileyLinkUnreachable
16:02:39.737 [Binder:9731_6] W/PUMPBTCOMM: [RileyLinkBLE$1.onConnectionStateChange():372]: RileyLink Disconnected.
16:02:39.783 [main] D/CORE: [DummyService.onStartCommand():70]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE)
16:02:39.813 [main] D/CORE: [DummyService.onStartCommand():70]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE)
16:02:40.817 [RxCachedThreadScheduler-23] D/AUTOMATION: [AutomationPlugin.onStart$lambda-9():140]: Grabbed new BT event: EventBTChange[deviceAddress=CF:6C:AA:F8:FC:BB,deviceName=OrangePro,state=DISCONNECT]
16:02:40.819 [RxCachedThreadScheduler-23] D/AUTOMATION: [AutomationPlugin.processActions():212]: processActions
16:02:40.820 [RxCachedThreadScheduler-23] D/AUTOMATION: [TriggerBg.shouldRun():63]: NOT ready for execution: Glucose is lesser than 4.0 MMOL
16:02:41.222 [main] E/PUMPBTCOMM: [RileyLinkBLEConfigActivity.startLeDeviceScan():198]: startLeDeviceScan failed: bleScanner is null
16:02:41.223 [main] D/PUMPBTCOMM: [RileyLinkBroadcastReceiver.onReceive():104]: Received Broadcast: AAPS.RileyLink.RileyLink_Disconnect
16:02:41.223 [main] I/PUMP: [RileyLinkServiceData.workWithServiceState():95]: RileyLink State Changed: BluetoothReady 
16:02:42.564 [main] E/PUMPBTCOMM: [RileyLinkBLEConfigActivity.startLeDeviceScan():198]: startLeDeviceScan failed: bleScanner is null
16:02:42.565 [main] D/PUMPBTCOMM: [RileyLinkBroadcastReceiver.onReceive():104]: Received Broadcast: AAPS.RileyLink.RileyLink_Disconnect
16:02:42.565 [main] I/PUMP: [RileyLinkServiceData.workWithServiceState():95]: RileyLink State Changed: BluetoothReady ```

happygirlnyc70, 2022-01-14:
Are you both using orange link v2.5? I had this issue as well but then flashed to the latest version 3.2 and it went away (pixel 4a 5g Android 12) https://getrileylink.org/orangelink-firmware

n0rbertg, 2022-01-14:
> Are you both using orange link v2.5? I had this issue as well but then flashed to the latest version 3.2, (there is a v3.2 for pro and non-pro), then it went away (pixel 4a 5g Android 12) https://getrileylink.org/orangelink-firmware

I have an Orangelink Pro with v3.2 (But as mentioned I tried it with Emalink and the original RileyLink as well and those didn't work either)

Jane49, 2022-01-14:
I flashed to 3.2 and then reflashed back to my original 2.5, it didn't make any difference for me.

Jane49, 2022-01-14:
> Are you both using orange link v2.5? I had this issue as well but then flashed to the latest version 3.2, (there is a v3.2 for pro and non-pro), then it went away (pixel 4a 5g Android 12) https://getrileylink.org/orangelink-firmware

I flashed to 3.2 and then reflashed back to my original 2.5, it didn't make any difference for me.

happygirlnyc70, 2022-01-14:
Hmmm missed that that you have other links.  I have emalink as well. Sorry that didn't fix it for you both

andyrozman, 2022-01-14:
We are assuming at the moment this has to do with new Android 12 BT permissions. There is similar issue opened #1173 . I will probably look into this over weekend.. My test phone (S20) is just upgrading to Android 12, so I will have hardware to test this.

MilosKozak, 2022-01-14:
fixed by https://github.com/nightscout/AndroidAPS/commit/e2bdc348349e6894c05eb0ba9809e378e10cb9d1

closed 2022-01-14 by MilosKozak

Jane49, 2022-01-14:
> We are assuming at the moment this has to do with new Android 12 BT permissions. There is similar issue opened #1173 . I will probably look into this over weekend.. My test phone (S20) is just upgrading to Android 12, so I will have hardware to test this.

Thank you :)

n0rbertg, 2022-01-14:
Thank you, I can confirm the fix works, scanning works as intended