nightscout/AndroidAPS #1167: Safety: increasing profile % without updating basal profile in pump !?

szantos, 2022-01-13:
It should be considered, whether a profile increase with duration should update the basal profile in the pump.

A safety issue can occur if the duration ends, and the basal profile can't be set back to lower levels (due to connection issues for example). The notification "Failed to update basal profile" helps a bit, but it won't stop administartion. If the phone is not by the user, the user is not even notified. If using automation for increasing profile the opportunity is higher for this scenario (even if 130% is the limit).

Regardless of the basal profile stored in the pump the algorithm could treat the increased basal profile as 100% for the loop calculations.

Also see:
[https://github.com/nightscout/AndroidAPS/issues/59](url)

MilosKozak, 2022-01-13:
If there is no connection to the pump loop cannot do anything as well

szantos, 2022-01-13:
Yes, and the increased! basal profile of the pump will run along and dose more than expected (and needed).

MilosKozak, 2022-01-13:
well this would not increase safety because you are only looking at single point (reverting to previous basal profile) and some time after. but people are running profile with duration for many hours too where wrong set profile would have much higher (and wrong) impact

closed 2022-01-13 by MilosKozak