nightscout/AndroidAPS #1169: 3.0rc1 SMB disabled

winni67, 2022-01-13:
Hello,

perhaps a new feature or possible a bug. I don‘t use my 3.0rc1 version for active looping. So I don‘t know, if the SMBs are disabled in 3.0rc1 or earlier in one of the beta versions.
When I started testing the 3.0 versions with beta1, then the SMBs are working.
But today, I saw this message (see picture). SMBs are enabled in the configuration.
I use xDrip+ (slave) for generating the glucose values on my AAPS3.0 test system.
Kind regards
[image]

MilosKozak, 2022-01-13:
what is on the top of this tab?

winni67, 2022-01-14:
[image]

parapenT1sta, 2022-01-14:
Do you have Red warning sign on the main screen regarding the BG readings?

winni67, 2022-01-14:
On my master system: xDrip+ with Dexcom G6 Native mode, xDrip+ master for followers. On my test system with AAPS 3.0: xDrip+ Sync Follower.
There is no red warning sign on the main screen of my AAPS3.0 test system.

winni67, 2022-01-14:
[image]
[image]
[image]
[image]

MilosKozak, 2022-01-14:
hmmm seems to be ok
then provide logs pls

overfrenk, 2022-01-15:
max minutes of basal to limit smb to 120 is very high,limit to max 90 for test.

overfrenk, 2022-01-15:
always to test : set safety multiplier to 5  both options and maximum total IOB to 25U.

winni67, 2022-01-15:
Logs send via mail.

winni67, 2022-01-16:
The mail address in AAPS 3.0 „SEND LOGS BY EMAIL“ xxx@androidaps.org can‘t be reached. The domain androidaps.org doesn‘t exist.

MilosKozak, 2022-01-16:
add it here directly

winni67, 2022-01-16:
[AndroidAPS_LOG_1642191610498.log.zip](https://github.com/nightscout/AndroidAPS/files/7877381/AndroidAPS_LOG_1642191610498.log.zip)

MilosKozak, 2022-01-16:
From what I see you are using BG source not supporting SMB always and you don't enter carbs. Thus COB is always zero and SMB is forced disabled

winni67, 2022-01-16:
Yes, that‘s already mentioned in the Constraints Part of the „OPENAPS SMB“ section. But when xDrip+ Sync Follower isn‘t allowed as BG source, then I can‘t test AAPS 3.0 on my test system anymore. The option to set a second G6 is possible, but then, I have to wear a lot of additional stuff close to me. Using it as active loops system is not my favorite.

closed 2022-01-16 by MilosKozak