nightscout/AndroidAPS #1174: [RC1] SMB/Close Loop stops after Calibrations

parapenT1sta, 2022-01-14:
There is a behaviour in AndroidAPS RC1 that I think should be reviewed. Everytime I calibrate using DexcomApp, and the value are very similar, AndroidAPS detects as two readings close to each other and stop the SMB/Close Loop.
In my opinion AndroidAPS should ignore "BG Check" values. This is happening to me quite often. I have to go manually remove the reading in red.
Calibration was at 8:59 and SMB/Close Loop stop working around 9:30 +/- (as you can check in the graph.

[AndroidAPS_LOG_1642156371451.log.zip](https://github.com/nightscout/AndroidAPS/files/7869732/AndroidAPS_LOG_1642156371451.log.zip)

[image]
[image]
[image]
[image]

MilosKozak, 2022-01-14:
time around 8:59 is not in logs

parapenT1sta, 2022-01-14:
I think these two logs cover the time frame of the events.
[AndroidAPS._2022-01-14_00-00-01_.14.zip](https://github.com/nightscout/AndroidAPS/files/7870659/AndroidAPS._2022-01-14_00-00-01_.14.zip)
[AndroidAPS._2022-01-14_00-00-01_.13.zip](https://github.com/nightscout/AndroidAPS/files/7870660/AndroidAPS._2022-01-14_00-00-01_.13.zip)

MilosKozak, 2022-01-14:
are you using G5?

MilosKozak, 2022-01-14:
it looks like a bug in Dexcom app. calibration is sent as BG value too

parapenT1sta, 2022-01-14:
Yes, G5.
As I wrote it seems AndroidAPS marks as "red" reading the value adjusted really after the calibration.

parapenT1sta, 2022-01-14:
@MilosKozak One note, the same problem happens when I initiate a new sensor and enter dual calibrations

parapenT1sta, 2022-01-14:
@MilosKozak One note, the same problem happens when I initiate a new sensor and enter dual calibrations
Same problem with xDrip or BYOD. AndroidAPS assume the dual calibrations as two readings at near time and marked them as red.

MilosKozak, 2022-01-14:
fixed by https://github.com/nightscout/AndroidAPS/commit/0603e741be6c6cd23d2d4b6c84c70de528b63a43
fixed for BOYDA. no way to fix it in xdrip on AAPS side

closed 2022-01-14 by MilosKozak

parapenT1sta, 2022-01-14:
Thank you @MilosKozak !

Regarding xDrip, what can be the solution? Changes on xDrip itself to avoid broadcast the calibrations?

MilosKozak, 2022-01-15:
i did only workaround as the issue is somewhere in G5 (maybe app only, who knows)
if you see the issue with xDrip too it must be analyzed there