nightscout/AndroidAPS #1177: [30 rc1] AAPS shows double down arrow at 283 and increasing delta value.

69nobody, 2022-01-14:
Although the blood glucose level is rising, two downward arrows are displayed in the AAPS interface.
In addition, I noticed that when entering carbohydrates, the message "Bolus cannot be delivered" is displayed and you can only stop the action.
An input via the bolus calculator with carbohydrates has worked.
Logs are attached.
[AndroidAPS._2022-01-14_22-20-06_.0.zip](https://github.com/nightscout/AndroidAPS/files/7873393/AndroidAPS._2022-01-14_22-20-06_.0.zip)

mick711, 2022-01-14:
I read last week same issue and if you see that it's a Xdrip problem .

69nobody, 2022-01-14:
thanks for this information. I will close my issue.

69nobody, 2022-01-14:
The same problem I described was reported by someone else a few days ago. This information was sent to me by mick711.

closed 2022-01-14 by 69nobody