nightscout/AndroidAPS #1178: [3.0 RC1] Separate entry of carbohydrates not possible.

69nobody, 2022-01-14:
When entering carbohydrates, the message "Bolus cannot be delivered" is displayed and you can only cancel the action.
No carbohydrates are then added.
An input via the bolus calculator with carbohydrates worked.
Protocols are attached.
[AndroidAPS._2022-01-14_22-20-06_.0.zip](https://github.com/nightscout/AndroidAPS/files/7873617/AndroidAPS._2022-01-14_22-20-06_.0.zip)

MilosKozak, 2022-01-15:
time of occurence?

69nobody, 2022-01-15:
In my memory, this must have happened in the evening. When I search my logs in AAPS I find only one entry with 25g carbohydrates at 20:58.

MilosKozak, 2022-01-16:
this time is not log file

69nobody, 2022-01-16:
Ok, sorry for my imprecise expression. I said that the input only for the carbohydrates did not work. However, the input via the bolus calculator worked. The input that worked was at 20:58. So the error occurred approximately between 20:50 and 20:58. There should be something in the log. If there was nothing to be found at this time, the behaviour was probably not recorded in the log.

MilosKozak, 2022-01-16:
log is from 21:45

69nobody, 2022-01-16:
Are the log files not continuously updated for one day?! Does the log file have to be exactly from 20:50 to 20:58? If the error cannot be found, then please take care of the other messages and close my issue.

MilosKozak, 2022-01-16:
you can open the file yourself and see the time

closed 2022-01-16 by MilosKozak