nightscout/AndroidAPS #1179: Duplicate CGM data as the readings , in virtual pump and data coming from Knighscout not via xdrip

g7kbr, 2022-01-15:
AndriodAPS 3.0-rc1

So running in virtual mode with a Medtronic 722.  The CGM is currently coming from my Knightscout account.  I am on objective 5.  I seem to have duplicate CGM data as the readings are very close multiple times.

I am almost at a stage to go Omnipod Dash, my CGM data is from a libre2.

I am a bit concerned about the red triangle.

Any feedback would be great.  I see that it is users on G6 dexcom which seem to see this.

Log file attached, and 3 x JPG's showing issue.  Using Build: 86ec769f6-2022.01.12-18:29

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in t
![IMG_7288](https://user-images.githubusercontent.com/27280641/149617894-ab584020-1801-46db-abb4-71554eb764b
[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/7874715/Log.Export.txt)
0.jpg)
he About dialog in the app, when pressing the three dots in the
[image]

[image]

  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

g7kbr, 2022-01-15:
> AndriodAPS 3.0-rc1
> 
> So running in virtual mode with a Medtronic 722. The CGM is currently coming from my Knightscout account. I am on objective 5. I seem to have duplicate CGM data as the readings are very close multiple times.
> 
> I am almost at a stage to go Omnipod Dash, my CGM data is from a libre2.
> 
> I am a bit concerned about the red triangle.
> 
> Any feedback would be great. I see that it is users on G6 dexcom which seem to see this.
> 
> Log file attached, and 3 x JPG's showing issue. Using Build: [86ec769](https://github.com/nightscout/AndroidAPS/commit/86ec769f6b1c711a9fd0e694681421f7c68858e3)-2022.01.12-18:29
> 
> ## Reporting bugs
> * **Note the precise time the problem occurred** and describe the circumstances and steps that caused
>   the problem
> * Note the Build version (found in t
>   ![IMG_7288](https://user-images.githubusercontent.com/27280641/149617894-ab584020-1801-46db-abb4-71554eb764b
>   [Log Export.txt](https://github.com/nightscout/AndroidAPS/files/7874715/Log.Export.txt)
>   0.jpg)
>   he About dialog in the app, when pressing the three dots in the
>   [image]
> 
> [image]
> 
> upper-right corner).
> 
> * Obtain the app's log files, which can be found on the phone in
>   _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
>   See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
> * Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
[image]
[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/7874819/Log.Export.txt)
[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/7874820/Log.Export.txt)

closed 2022-01-15 by g7kbr

g7kbr67, 2022-01-15:
Re-opened as I didn't realise I closed it.