nightscout/AndroidAPS #1180: Duplicate CGM data as the readings , in virtual pump and data coming from Knighscout not via xdrip

g7kbr, 2022-01-15:
**AndriodAPS 3.0-rc1

So running in virtual mode with a Medtronic 722. The CGM is currently coming from my Knightscout account. I am on objective 5. I seem to have duplicate CGM data as the readings are very close multiple times.

I am almost at a stage to go Omnipod Dash, my CGM data is from a libre2. Tis I will use OOP2 and xdrip.

I am a bit concerned about the red triangle.  As I am not using xdrip at the moment.

Any feedback would be great. I see that it is users on G6 dexcom which seem to see this.**

**Attached Logs and screenshots JPG's below**

[Log.Export.txt](https://github.com/nightscout/AndroidAPS/files/7874971/Log.Export.txt)
[image]
[image]
[image]
[AndroidAPS._2022-01-14_16-32-39_.0[1].zip](https://github.com/nightscout/AndroidAPS/files/7874973/AndroidAPS._2022-01-14_16-32-39_.0.1.zip)
[AndroidAPS._2022-01-15_00-00-16_.0[1].zip](https://github.com/nightscout/AndroidAPS/files/7874974/AndroidAPS._2022-01-15_00-00-16_.0.1.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
[Log.Export.txt](https://github.com/nightscout/AndroidAPS/files/7874971/Log.Export.txt)
[image]
[image]
[image]
[AndroidAPS._2022-01-14_16-32-39_.0[1].zip](https://github.com/nightscout/AndroidAPS/files/7874973/AndroidAPS._2022-01-14_16-32-39_.0.1.zip)
[AndroidAPS._2022-01-15_00-00-16_.0[1].zip](https://github.com/nightscout/AndroidAPS/files/7874974/AndroidAPS._2022-01-15_00-00-16_.0.1.zip)

andyrozman, 2022-01-15:
Are you using Xdrip? If yes at the moment Xdrip is not compatible with AAPS 3.0... Xdrip team is working on fixing the problem, so I am hoping this will be possible soon.

g7kbr67, 2022-01-15:
No I am not using Xdrip, my current CGM data is via my iPhone and that I am using xDrip4iOS via freeaps.  AndroidAPS 3.0 is getting its data via Knightscout.

MilosKozak, 2022-01-15:
AAPS is just telling you your bg source has duplicated readings. it's not a bug of AAPS

andyrozman, 2022-01-15:
if xDrip4iOS is using the same base code as Xdrip it might have the same problem... If wrong data is saved into NS, and AAPS reads from NS, then it gets wrong data...
And like Milos said, it looks like you have duplicated readings...

g7kbr67, 2022-01-15:
> duplicated

So would I need to manually delete the duplicate reading, frequently? I am not closed loop yet but am I write in saying closed loop would stop if duplicates are found?

MilosKozak, 2022-01-16:
you have to resolve the issue on your BG source. Using NS BG source is last resort. you should send BG directly to AAPS to avoid necessity of internet connection
And yes, duplicated BGs leads to stoped loop to prevent errors in calculations

closed 2022-01-16 by MilosKozak