nightscout/AndroidAPS #1182: Strange Omnipod DASH error code

christinadamianou, 2022-01-15:
While injecting a bolus with Omnipod DASH, I received a POD Fault: 49.
Officiali Insulet error codes do not include 049. What does it stands for?
POD was beeping until deactivated.

AndroidAPS 3.0-beta11
build: e39014962-2021.12.19-13:34

Log Files:
[AndroidAPS._2022-01-05_13-15-23_.0.zip](https://github.com/nightscout/AndroidAPS/files/7875544/AndroidAPS._2022-01-05_13-15-23_.0.zip)
[AndroidAPS._2022-01-05_13-15-23_.1.zip](https://github.com/nightscout/AndroidAPS/files/7875545/AndroidAPS._2022-01-05_13-15-23_.1.zip)
[image]

avereha, 2022-01-17:
Please share all the steps that you tried and as many details as possible.

From what I can see in the logs:
 - you used the wizard to request a 5.65 bolus. This started to get delivered.
 - the app crashed while the bolus was delivered. Or did you close it on purpose?
 - you requested another bolus of 5.6 while the previous bolus was being delivered. Was this on purpose. did you want to request 11U or did you think that the previous bolus failed?
 
I could add a check for this case, but maybe it's better in this case to let the pod fail to prevent people from issuing multiple boluses.

christinadamianou, 2022-01-17:
I requested 5.65 bolus. App seemed to stuck
I killed the app and requested new bolus, because I didnt receive any notification that insulin was injected.

It sounds ok the way the app acts when POD fails.
My question is that Insulet officially does not give the 049 error code, so what does it means? What would be more helpful is to have the full error code (digits dash digits dash digits) somewhere available.

avereha, 2022-01-18:
This might help: https://loopkit.github.io/loopdocs/troubleshooting/omnipod-faults/

avereha, 2022-01-21:
Fixed with https://github.com/nightscout/AndroidAPS/pull/1211

closed 2022-01-22 by MilosKozak