nightscout/AndroidAPS #1183: SNOOZE text inside the snooze button is not responsive

dar-kod, 2022-01-15:
AndroidAPS 3.0-rc1
Build version: 86ec769f6
Phone: Poco F3
Settings -> Display -> Text size: L

Maybe this is a phone specific, but I find this not optimized:

[image]

MilosKozak, 2022-01-16:
AAPS is not so optimize for very small screens. if you rotate to landscape is it OK?

dar-kod, 2022-01-17:
If I rotate to landscape, it is the same.
[image]

nicoleloop, 2022-01-21:
The snooze option has never worked for me in the 2.5 years I have used AndroidAPS. Has it worked for everyone else all this time?

dar-kod, 2022-01-22:
@MilosKozak, the issue in my case is layout. Is it possible to fix it?

MilosKozak, 2022-01-22:
improved by https://github.com/nightscout/AndroidAPS/commit/72e73f604f124ea08d3dc0a06a6765dbe9fe4899

closed 2022-01-22 by MilosKozak