nightscout/AndroidAPS #1184: AndroidAPS doesn't take BG value in main window

dar-kod, 2022-01-15:
AndroidAPS 3.0-rc1
Build version: 86ec769f6
Phone: Poco F3
BG Source: xDrip+ (build 73ada42)
Time: 18:37

It is not a first time that I see such behavior. I have to see is there some rule for this to appear. In the screenshots, you can see that the BG value in the home window or tab is crossed while at the same time xDrip+ tab and xDrip+ app show the current BG reading.
[image]
[image]
[image]
[AndroidAPS._2022-01-12_14-11-14_.1.zip](https://github.com/nightscout/AndroidAPS/files/7876074/AndroidAPS._2022-01-12_14-11-14_.1.zip)

MilosKozak, 2022-01-16:
Enable EVENT logging and provide logs if you see this issue again

dar-kod, 2022-01-16:
Thank you. I have enabled EVENT logging now.

dar-kod, 2022-01-17:
It just happened again at 9:02. I'm attaching screenshots and a log file at the end
[image]
[image]
![photo_2022-01-17_09-44-45](https://user-imag
[AndroidAPS._2022-01-17_00-00-05_.11.zip](https://github.com/nightscout/AndroidAPS/files/7880418/AndroidAPS._2022-01-17_00-00-05_.11.zip)
es.githubusercontent.com/24446806/149738007-c0bfd7af-27f0-4b9a-9ddd-2dc95801cea5.jpg)

.

MilosKozak, 2022-01-20:
@regiorunner next time pls keep phone time on screenshot
anyway AAPS didn't receive any data from xdrip for 15 min
```
14:46:06.553 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -4.2391448032144006E-5; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2110041402 49817a5-2021.10.04; com.eveningoutpost.dexdrip.Extras.Raw => 93.3020433770585; com.eveningoutpost.dexdrip.Extras.Time => 1641822365657; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => SingleDown; com.eveningoutpost.dexdrip.Extras.BgEstimate => 93.3020433770585; com.eveningoutpost.dexdrip.Extras.SourceDesc => Tomato; }Bundle
14:56:15.257 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -4.566866909977561E-5; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2110041402 49817a5-2021.10.04; com.eveningoutpost.dexdrip.Extras.Raw => 65.4983182677477; com.eveningoutpost.dexdrip.Extras.Time => 1641822974471; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => SingleDown; com.eveningoutpost.dexdrip.Extras.BgEstimate => 65.4983182677477; com.eveningoutpost.dexdrip.Extras.SourceDesc => Tomato; }Bundle
15:01:15.122 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -1.6007960682753334E-5; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2110041402 49817a5-2021.10.04; com.eveningoutpost.dexdrip.Extras.Raw => 60.701196681986325; com.eveningoutpost.dexdrip.Extras.Time => 1641823274142; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 60.70119668198633; com.eveningoutpost.dexdrip.Extras.SourceDesc => Tomato; }Bundle
15:16:29.361 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -2.7996618074964855E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2110041402 49817a5-2021.10.04; com.eveningoutpost.dexdrip.Extras.Raw => 59.820092717254646; com.eveningoutpost.dexdrip.Extras.Time => 1641823588860; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 59.82009271725465; com.eveningoutpost.dexdrip.Extras.SourceDesc => Tomato; }Bundle
15:16:29.488 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -6.526696035049421E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2110041402 49817a5-2021.10.04; com.eveningoutpost.dexdrip.Extras.Raw => 57.86208390673983; com.eveningoutpost.dexdrip.Extras.Time => 1641823888860; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 57.86208390673983; com.eveningoutpost.dexdrip.Extras.SourceDesc => Tomato; }Bundle
15:16:29.607 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -4.242352422782195E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2110041402 49817a5-2021.10.04; com.eveningoutpost.dexdrip.Extras.Raw => 56.58937817990517; com.eveningoutpost.dexdrip.Extras.Time => 1641824188860; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip. [... cut ...] ras.Time => 1641825688009; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1640809446350; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => 80; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 78.32327597661994; com.eveningoutpost.dexdrip.Extras.SourceDesc => Tomato; }Bundle
```

MilosKozak, 2022-01-20:
and if you look into logs you'll see phone was in really strange state 15.00-15.15

MilosKozak, 2022-01-20:
@dar-kod tha same is valid for your phone
no data from xdrip 8:37-9:02
```
08:32:19.810 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -9.99796708002706E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2122011519 c25c5dc-2022.01.15; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1642404736613; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 87.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G6 Native; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
08:37:19.683 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => 3.334133525379425E-5; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2122011519 c25c5dc-2022.01.15; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1642405036541; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => SingleUp; com.eveningoutpost.dexdrip.Extras.BgEstimate => 97.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G6 Native; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
09:02:19.791 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => 2.6665937797700194E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2122011519 c25c5dc-2022.01.15; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1642406536582; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 101.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G6 Native; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
09:02:23.662 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => 6.6666444445185185E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2122011519 c25c5dc-2022.01.15; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1642405336581; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 105.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G6 Native::Backfill; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
09:02:23.692 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => 6.6666444445185185E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2122011519 c25c5dc-2022.01.15; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1642405936581; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 100.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G6 Native::Backfill; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
```
it's not fault of AAPS but something is probably wrong with your phone, prmissions, other installed apps etc

dar-kod, 2022-01-22:
@MilosKozak, you wrote: "the same is valid for your phone no data from xdrip 8:37-9:02"

Does it mean that data was received at 9:02 or 9:07? At 9:06 data was there in XDRP+ tab but not in the home tab or main screen of AAPS. Did AAPS receive missing readings at 9:02? If yes, then it is AAPS issue.

I didn't mention that "no data" happens because I was away from the phone for, in this case, about 20 minutes.

MilosKozak, 2022-01-22:
i probably don't undestand. you were away from phone and expecting there will be BGs?

dar-kod, 2022-01-22:
Not like that. I expected this: as soon as xDrip+ has BGs, they will appear also in AAPS but acording to what I see without logs BGs appeared only in AAPS **XDRIP+** tab.

dar-kod, 2022-01-28:
@MilosKozak, did you read my previous answer? I think that this issue is wrongly closed. 

What I didn't mention at the beginning is that this happens only if I am away of my phone for some time and no readings come to xDrip+ and AAPS. But as soon as I come to my phone and the missing reading come to xDrip+, the readings appear not all in AAPS main screen but only in XDRIP+ tab and are not taken into AAPS calculations. **After another 5 minutes of waiting, the all readings come** in the main or home screen of AAPS. **I would expect that all reading come to AAPS main screen as soon as they come to xDrip+ and XDRIP+ tab in AAPS.**

From my point of view, this is AAPS issue.

MilosKozak, 2022-01-29:
@dar-kod provide new logs with GLUCOSE looging enabled

dar-kod, 2022-02-04:
I am providing new logs with GLUCOSE enabled:
[image]
[image]
[image]
[AndroidAPS._2022-02-04_00-00-00_.20.zip](https://github.com/nightscout/AndroidAPS/files/8004586/AndroidAPS._2022-02-04_00-00-00_.20.zip)

eversteegt, 2022-06-01:
We have an issue which is probably related (will enable logging when I get my hands on my GF's phone again, since she is the user of AAPS ).  It looks like it sort of started either after the upgrade to AAPS 3.x, or a recent Android security update from Samsung (A52-5g). 

Sensor: Libre 2, either with native Xdrip BT connection (and OOP2) or with Miaomiao2 (tomato) transmitter. 

ALL permissions for AAPS, OOP2 and Xdrip are enabled and Xdrip's inter-app communication settings point toward AAPS. Also, the apps are exempted from any battery optimization. 

Aside from the ever permanently  present yellow triangle (due to the irregular nature of Libre readings), AAPS, at random moments, starts screaming no BG data was received for some long (extended) period of time, sometimes more than one hour. That message is totally incorrect, as AAPS has a complete BG graph and, before the error/alarm popped up, definitely had actual (recent) value.

duncanbarnes, 2023-01-22:
Looks the same as #2004 so fixed by https://github.com/nightscout/AndroidAPS/commit/8fb5c5a984a8e16062e70282ce3d0fd6383ed864

?