nightscout/AndroidAPS #1186: DASH - Request to partially turn off the pod expiration reminder

dar-kod, 2022-01-16:
AndroidAPS 3.0-rc1
Build version: 86ec769f6

In Dash (Settings) -> Alerts -> Expiration reminder there is only the option to completely turn off the reminder for the expiration of the pod. Generally there are two reminders, one before 72 hours and the other before 80 hours. If I remember correctly, in Eros it was possible to set each reminder separately. For those who use the POD for 80 hours and want to have an alarm, two alarms are too much. In short, if possible, it would be desirable to have the option to set and turn off for both reminders.

regiorunner, 2022-02-01:
I am also interested in this feature to turn off the alarm for 72 hours  for the Dash. I think the pod alarm at 79 hours (one hour before the pod really expires) cannot be turned off for safety reasons. But there might be something different for the Dash compared to the Eros. I disabled the alarm for expirition, but the Dash pod beeps at exact 72 hours. This might be an additional built-in safety alarm. I will test it by enabling the expiration alarm at 71 hours and see whether I get two alarms, one at 71 hours and one at 72 hours. Let you know in three days.

regiorunner, 2022-02-07:
I confirm it is as I expected in my previous comment.
Enabling the alarm for expiry of the Dash indeed causes the Dash pod to sound an alarm at the adjusted hours before the expiry of the pod. The reference for the hours before expiry of the Dash is 72 hours, this was 80 hours for the Eros.
At 72 hours exactly the Dash sounds an additional alarm. This alarm was not present in the Eros.
At 79 hours (one hour before real expiration) the Dash sounds an alarm, this is the same as the Eros. 

Summary to enhance the alarm control for tyhe Dash:
- Disabling the additional alarm of the Dash xx hours before expiry is already implemented.
   It would be nice to refer the xx hours to the 80 hrs expiry instead of the 72 hours, like the Eros.
- The 72 hours alarm of the Dash is an additional alarm, this should be seperately disabled (if possible).

I hope this clarifies the sutuation.

rustymonkey, 2022-03-17:
fyi, there's similar feedback about this issue in this [thread in the AAPS FB group](https://www.facebook.com/groups/AndroidAPSUsers/posts/3212159959005399/?__cft__[0]=AZWL8XkAm7f-DvHOma6TTpCDCp850WdpTKzFpX0NDzQCNPJGM01EcvBv1WsH5Jtx-gphccP7yrKtDGt_bHma0gy2-Qq5n7VtdW6JXzYDvWk-uQuQDWbZcNoJnFBVz999A4rpadmG-9zYK-yRsVdufYWD&__tn__=%2CO%2CP-R).

olorinmaia, 2023-09-15:
This issue can be closed, as it has been fixed in: https://github.com/nightscout/AndroidAPS/issues/2691
https://github.com/nightscout/AndroidAPS/commit/d6e6a986427c8ff9db464e0dee92639fc526e94e

parapenT1sta, 2023-09-16:
@MilosKozak I think you can close this issue.

olorinmaia, 2023-10-31:
@MilosKozak You can close this issue :) https://github.com/nightscout/AndroidAPS/commit/d6e6a986427c8ff9db464e0dee92639fc526e94e

closed 2023-10-31 by MilosKozak