nightscout/AndroidAPS #1187: Feature Request: Insulin left estimate for greater than 50U in Omnipod Dash (3.0 RC1)

stevensonderekr, 2022-01-16:
It would be very nice to be able to estimate how much insulin I have in my Dash pod over the 50 unit mark. If there was an optional prompt during the pod change for approximate amount filled, that coupled with the total amount delivered by the pod could give a reasonable estimate of the insulin left until the pod itself is able to report the value.

Freloner, 2022-01-16:
I think this is hardware related, maybe @avereha could confirm

andyrozman, 2022-01-16:
Yes it is hardware related. 

Pod doesn't give details about how much insulin is left, until it reaches 50 U. 

There is nothing to do here.

stevensonderekr, 2022-01-16:
It's definitely a limit of the pod hardware that it can only detect the amount of insulin in the reservoir once it's under 50u. I'm certain that it could be worked around with known information since it would be nice to have an estimated value on the AAPS overview at a glance. Currently I need to do mental math (200u filled minus insulin delivered) looking at the Dash status screen, and I'll often forget to. Sometimes I need to know if I have more than say, 80-100 units in my reservoir to plan for how many supplies I need to keep on my person versus in a bag/locker/etc, and this would help to make that easier. It's one of the things I really miss from other pumps I've used in the past.

avereha, 2022-01-17:
Correct, the pod reports only when it has less than 50U.

kristinsmotz, 2022-01-18:
This goes for both Eros and Dash, hardware limitation is that the pod only reports below 50U. I agree it would be nice to have estimates above that. 

Adding a "filled amount" input in Activate Pod dialog, or even as a separate button in Pod Management (or config setting for usual fill amount), and a calculated estimate with "filled amount" minus "total delivered" = "estimated remaining" would be a way to do that. 

I am not sure I would want it to override the predefined ">50U" in the current field though, but perhaps a single use extra field in the POD/DASH tab would be helpful. That way there would be no need to mess with current logic on Pump/Remaining field, at least for now. If it works well and users report liking it, it could be a possible future upgrade to add the logic to Pump/Remaining (which would also render the info to NS / NSClient for parent monitoring).

crizz6396, 2022-02-09:
Fyi, a pod can be overfilled. The pump will deliver until it is empty.  I noticed that aaps is assuming there is less than 200 because of the prime. Is this an aaps calculation or pump calculation?

avereha, 2022-02-28:
What problem are we trying to solve here with this feature?

stevensonderekr, 2022-02-28:
[image]
This is an issue for me because I do not know how much insulin over 50U I have on board, and I need to know that quickly before leaving my home. I have to plan differently for my day if I have 80U on board versus, say, 50U exactly.
I want it to read "170U EST" since that's a ballpark of what I have in the pod, or at least "50U+"
This should be achievable by prompting the user for an estimated fill amount when the pod is initialized, or afterwards could be manually entered if changing that process is not desired.
The value on the overview screen can revert to what the pod reports when under 50U remain.

crizz6396, 2022-03-01:
If you go to this screen you can get a better estimate. But remember it's
just an estimate

mick711, 2022-05-12:
https://github.com/nightscout/AndroidAPS/commit/caa1f18c5ac25b4e2c7377df1193a02705a98ae2

kristinsmotz, 2022-05-12:
How does this commit help, and why on earth was it approved into dev? It only sets a different and potentially even more wrong value (75) than just displaying "over 50U". Now we could be fooled to think there are 75U left until there is only 50U. Then when below it would jump to actual pod estimated remaining amount, so in minutes go from "75U" to 49U.  This commit in no way calculates "fill amount minus total delivered" like the feature request is asking for, which would actually be helpful.

MilosKozak, 2026-10-15:
now AAPS shows amount of used insulin per reservoir

closed 2026-10-15 by MilosKozak