nightscout/AndroidAPS #1188: Improve UI-interaction to change viewrange on overview (feature request)

vanelsberg, 2022-01-16:
**Improve UI-interaction to change view range on overview (feature request):**

**Proposed solution options**
I can think of several ways to make changing the view more user friendly:
1. UI option on overview (popup/dropdown?) to directly select on of the 4 possible view ranges.
2. Single UI action for the user to go back to a configurable default view range .
3. UI automatically switching to user configurable default after NN minutes?

(Personally I would prefer option 1)

**Reason:**
As changing the view range obviously takes time it can never be very responsive. Mostly because of this I have grown to dislike the UI interaction method to change the view range (tab-hold-and-wait for rendering).
As is now I often find myself hunting for the desired view range: tab-hold-wait, tab-hold-wait, Oops... one to many?
Go round again: tap-hold-etc.... Quite tedious sometimes!

Philoul, 2022-01-18:
What do you think of this proposal in PR #1196 (I used existing button with a long click to show rescale menu)
=> Long press on graph remains unchanged
=> Short click on button remains for sub graphs selection
=> Long press on button allow a direct selection of scale (between 6 hours and 24 hours) (in french in my screenshot)

[image]

Andries-Smit, 2022-01-18:
**A side note;** The `Long press` pattern is not ideal for the learning experience of AAPS, a user will not try random long press on elements, making slowing down the learning curve. Yesterday I found trough browsing through the code, there are move places. Ideal for shortcuts for power users, but new users need an good alternative trough self-explaining user interface.

Philoul, 2022-01-18:
@Andries-Smit What do you suggest here? (additional button ? something else?)

vanelsberg, 2022-01-18:
@Philoul Agree with Andries on the long press pattern.
Also bit worried that could be affected by unresponsiveness while rescaling?

**Yes**. In general: I do think the direct scale selection would be an improvement.
Problem is: there is no space for another button. So maybe see how short/long press works out on usage?

vanelsberg, 2022-01-18:
@Philoul 
Personally I would love to have a simple way to get back to a (user configurable) default view range without the need to go to a menu. In general I change the view range only when looking for some "event", then switch back to the 6-hour view.

Just dropping some ideas:
Any possibility to use double-click on the overview?
Or maybe switch to the default view range after some time threshold?

vanelsberg, 2022-01-18:
Adding: all details mentioned earlier: nice-to-have?

For me: if we could have the direct viewrange menu that would be a real improvement!
So lets keep this simple to start with?

Philoul, 2022-01-18:
I also restarted to work on PR #911 to (try to) find a solution that works for both overview and history browser 🤯

Philoul, 2022-01-24:
Hi,
If you want to test some proposals and give me a feedback, I made 3 PR
PR #1196 is independant and can be merged with PR #911 or with PR #1142
but currently if you try to merge PR #911 and #1142 together you'll get conflicts because same lines are modified in both PR
=> I can make another one with all proposals together for testing (not done because I prefer working on each subject separatly)

vanelsberg, 2022-01-24:
@Philoul Think I can do that this week.
Will test 1196+911 and 1196+1142 then. Anything particular to test for?

Philoul, 2022-01-24:
for #911, on my side graph, rescale is quicker, but I still have sometime the "application not responding" popup, especially just after a new BG is received... (a bit less popup than with current dev but I'm not sure).
You can also test if history browser works well for all PR.
thx

vanelsberg, 2022-01-24:
> for #911
Think I can do some testen test by wednesday 26/01. Will let you know my findings.

vanelsberg, 2023-08-20:
Solved for current v3.2 UI...

closed 2023-08-20 by vanelsberg