nightscout/AndroidAPS #1189: Android Studio error when trying to build 3.0 betas or 3.0rc1/rc2

FreDiabetics, 2022-01-16:
When trying to build any 3.0 beta or 3.0 rc I'm getting the following error. Master can be build without any issues.
I already deleted the whole project and cloned it again with the same result. anybody else with this?

Project 'omnipod' is ambiguous in root project 'AndroidAPS'. Candidates are: 'omnipod-common', 'omnipod-dash', 'omnipod-eros'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

[image]

MilosKozak, 2022-01-17:
you may try this 
[image]

MilosKozak, 2022-01-20:
closed for inactivity