nightscout/AndroidAPS #1190: V3 RC2 Bug : Message new version available

robertrub, 2022-01-17:
The RC1 bug is still there in RC2. There is a message telling that there is a new version since 1 day and closed loop will stop on a week etc.

There is no newer version than RC2...

[image]

Alamo04, 2022-01-17:
> The RC1 bug is still there in RC2. There is a message telling that there is a new version since 1 day and closed loop will stop in a week and 14 days etc.
> 
> There is no newer version than RC2...
> 
> [image]

Are you sure that this is RC2 ? Perhaps you installed RC1 again ?
Don't see that message with RC2.

robertrub, 2022-01-17:
I have checked and I have RC2 installed.

[image]

happygirlnyc70, 2022-01-17:
I can confirm, I also got this message rc2 
[image]

Alamo04, 2022-01-17:
Confirmed,  just now I become this message too.
Version runs 16h 45 minutes now.

robertrub, 2022-01-18:
Same issue persists on RC3 too.

[image]
[image]

happygirlnyc70, 2022-01-18:
The error is still present for rc3
[image]

sboernemeyer, 2022-01-20:
CONFIRM
also in RC3

catherinefranq, 2022-01-20:
oui egalement pour moi en rc3
[image]

nicoleloop, 2022-01-21:
Oh good I thought I was losing my mind lol. Yes, every night for the past 2-3 nights in a row at exactly 1am the same message appears on my phone's app as well. Why only at 1am? No clue, but it is an annoying time for it to appear...

robertrub, 2022-01-21:
Mine is around 10AM (GMT+1). Probably depends on you time zone.

MilosKozak, 2022-01-21:
once in 24h. depends on when the last check was done

nicoleloop, 2022-01-21:
Oh, of course! Thanks for the quick response Milos. 🙂

closed 2022-03-07 by robertrub