nightscout/AndroidAPS #1194: Silent Crash of AAPS

Futaba12, 2022-01-18:
Reporting bugs
--------------
Setup : 
- Phone One plus Nord with Android 11
- AAPS version : 2.8.2.19-dev (Build 0a1cbdb91)
- All battery optimization are deactivated
- BG from BYODA only in version 1.7.03

AAPS has crash silently. The first screenshot shows around 3pm30 (17 january 2021) that AAPS stops everything and the most problematic thing is that the pod doesn't fallback on my basal profile until I check my phone around 6pm. The second screenshot shows the pod history ans we can see a TBR set to 0u at 3pm24 precisely and another TBR set at 3.2u at 6pm14 when I check my phone. In this time frame, my bg has rised (Look at the last screenshot in red rectangle).

I also attach the AAPS logs just after the issue observed.

[AndroidAPS_LOG_1642440166854.log.zip](https://github.com/nightscout/AndroidAPS/files/7888178/AndroidAPS_LOG_1642440166854.log.zip)
[image]
[image]
[image]

MilosKozak, 2022-01-18:
sorry, reason is not visible in logs

closed 2022-01-18 by MilosKozak

Philoul, 2022-01-18:
@Futaba12 do you have unchecked items in "Maintenance/Paramètres journal" (in french ;-) ) ?
=> If yes, can you "enable" all items (to get maximum informations in log file and help us understanding what's your problem?
We cannot see any useful information in your log to be able to understand what could be the root cause of your crash...

ToreBj, 2022-01-18:
[image]
Have you disabled these?

Futaba12, 2022-01-18:
> @Futaba12 do you have unchecked items in "Maintenance/Paramètres journal" (in french ;-) ) ?
> => If yes, can you "enable" all items (to get maximum informations in log file and help us understanding [...]

Here a screenshot of the log parameters :

[image]